Avnu/OpenAvnu #373: AVB supported Endpoint

insanewifi, 2016-04-15:
Is there any difference between standard Ethernet NIC and AVB NIC. Can I use a any standard ethernet as AVB end point to stream only one channel of audio or video? Can i make 2 beaglebone as talker and listner. I have a AVB switch from broadcom.
Thank you

intel-ethernet, 2016-04-15:
Dear Insanewifi –

It depends.

To receive a stream, you can use any standard Ethernet controller with packet time-stamping support in the Ethernet controller to support gPTP (IEEE 802.1AS). You need to support gPTP at a minimum as unless a port is ‘asCapable’ the switch won’t stream AVB data.

To send a stream, you need packet time-stamping + at least the ability to rate limit the bursts of Ethernet traffic to never exceed the allocated bandwidth for the stream measured in “packets x packet_size / observation interval”. E.g. using the stream reservation protocol (SRP) bandwidth is allocated for lets say 1 packet x 200 bytes every 250 usec.

Nothing *requires* you to actually use this much bandwidth, although purists will say you are wasting available bandwidth if you over-allocate - for professional installations, they need to maximize the capacity of the network. That being said, you can implement a pure software-based scheduler to pace out the traffic if necessary, although in reality/practice live audio samples only get generated at a certain rate anyway. In this case you can just stream the packets out at the rate the data is produced, with the above restrictions that you reserve the appropriate amount of bandwidth and have some method to verify you aren’t exceeding the allocated bandwidth (with pacing as necessary).

For maximum bandwidth efficiency and utilization, you need an Ethernet controller which implements an ‘FTQSS’ (Qav) credit-based shaper, or time-synchronized packet transmission. If you want AVnu certification (which tests effectively how well behaved an implementation is and its interoperability with other devices), you need to have a controller with such features. The Intel I210 is one such adapter, although there are others which meet the requirements as well.

Hope this helps
ekm

intel-ethernet, 2016-04-15:
s/FTQSS/FQTSS/

link here - http://www.ieee802.org/1/pages/802.1av.html

insanewifi, 2016-04-18:
Thank you for the reply.

closed 2016-04-18 by insanewifi