Avnu/OpenAvnu #379: FollowUP TLV error

exustheone, 2016-04-26:
I have one question about some part of the code. On file:
https://github.com/AVnu/Open-AVB/blob/master/daemons/gptp/common/ptp_message.cpp

In line 226:
`memcpy( &(followup_msg->tlv), buf+PTP_FOLLOWUP_LENGTH, sizeof(followup_msg->tlv) );`

When copying the TLV frame, it should add the PTP header size to the calculation, not only the Follow Up length. With this, the follow up is being copied from the middle of the ptp header instead from the actual TLV part. So, it should be:

`memcpy( &(followup_msg->tlv), buf + PTP_COMMON_HDR_LENGTH + PTP_FOLLOWUP_LENGTH, sizeof(followup_msg->tlv) );`

andrew-elder, 2016-04-26:
Looks like a bug to me. Do you wish to submit a pull request with the fix to open-avb-next?

exustheone, 2016-04-26:
If you are asking me, i can't really push from where i work. So, i will need somebody else to actually push this fix.

And, in the same topic.
In file: 
https://github.com/AVnu/Open-AVB/blob/feature-gptp-next/daemons/gptp/common/avbts_message.hpp#658

Is defined the attribute:
`int32_t scaledLastGmFreqChange;`

Which from what i understand, it's saved with network endianness. Now, if we look at the getter and setter in here:
https://github.com/AVnu/Open-AVB/blob/feature-gptp-next/daemons/gptp/common/avbts_message.hpp#L697-L709

`void setScaledLastGmFreqChange(int32_t val)
    {
        scaledLastGmFreqChange = PLAT_htonl(val);
    }`

`int32_t getScaledLastGmFreqChange(void)
    {
        return scaledLastGmFreqChange;
    }`

You will see that the setter expects host endianness, while the getter returns the attribute as it is, meaning in network endianness.
But in line:
https://github.com/AVnu/Open-AVB/blob/feature-gptp-next/daemons/gptp/common/avbts_message.hpp#L817

` tlv.setScaledLastGmFreqChange(fup->getScaledLastGmFreqChange());`

we get the value in network endianness, and give it to the setter which expects host endianness. And this will apply the conversion to network endianness(PLAT_htonl). In case host and network have the same endianness, this will be ok, but in case they don't, it will swap again the bytes.

So, am i seeing this wrong(which is totally possible) or it's also a bug?

andrew-elder, 2016-04-26:
Please create a separate issue for a new discussion topic/bug.

andrew-elder, 2016-04-26:
Any gPTP experts have a comment on this issue?

christopher-s-hall, 2016-04-26:
This sounds like the right thing. I would rather see PTP_FOLLOWUP_OFFSET rather than PTP_COMMON_HDR_LENGTH for consistency.

andrew-elder, 2016-04-26:
@christopher-s-hall - do you want to go ahead and make the change? Sounds like you have it clear in your head what the preferred change would be.

pinealservo, 2016-05-17:
I believe this was fixed in pull request #394, which has been merged to open-avb-next.

closed 2016-05-17 by pinealservo