Avnu/OpenAvnu #380: FollowUP TLV endianess error

exustheone, 2016-04-26:
In file: 
https://github.com/AVnu/Open-AVB/blob/feature-gptp-next/daemons/gptp/common/avbts_message.hpp#658

Is defined the attribute:
`int32_t scaledLastGmFreqChange;`

Which from what i understand, it's saved with network endianness. Now, if we look at the getter and setter in here:
https://github.com/AVnu/Open-AVB/blob/feature-gptp-next/daemons/gptp/common/avbts_message.hpp#L697-L709

`void setScaledLastGmFreqChange(int32_t val)
    {
        scaledLastGmFreqChange = PLAT_htonl(val);
    }`

`int32_t getScaledLastGmFreqChange(void)
    {
        return scaledLastGmFreqChange;
    }`

You will see that the setter expects host endianness, while the getter returns the attribute as it is, meaning in network endianness.
But in line:
https://github.com/AVnu/Open-AVB/blob/feature-gptp-next/daemons/gptp/common/avbts_message.hpp#L817

` tlv.setScaledLastGmFreqChange(fup->getScaledLastGmFreqChange());`

we get the value in network endianness, and give it to the setter which expects host endianness. And this will apply the conversion to network endianness(PLAT_htonl). In case host and network have the same endianness, this will be ok, but in case they don't, it will swap again the bytes.

So, am i seeing this wrong(which is totally possible) or it's also a bug?

andrew-elder, 2016-09-22:
Does anyone know if this issue has been fixed?

pinealservo, 2016-09-23:
Yes; those links are to an old branch. The problem referenced doesn't exist in open-avb-next. I didn't check master, but I'm fairly sure it was fixed before the last merge to master.

closed 2016-09-23 by pinealservo