Avnu/OpenAvnu #382: "MSRP:Empty" response from SRP daemon

DMats, 2016-04-30:
I'm attempting to run the echo talker and echo listener example using two linux computers with I210 cards installed.

After downloading the Open-AVB repo, I run the following at the top level from an admin terminal.  My only edit is to change the run_echo_talker.sh and run_echo_listener.sh "stream_addr" argument to the MAC of the "talker" linux machine.
```
make igb
ARCH=I210 make daemons_all
make avtp_pipeline
./run_igb.sh eth1
./run_gptp.sh eth1
./run_srp.sh eth1
./run_echo_listener eth1
./run_echo_talker eth1

```
However, the SRP fails to initialize on both the talker and listener sides.  Like the following:
```
[user@localhost Open-AVB-master]$ sudo ./run_echo_talker.sh enp1s0
attaching to 0000:02:00.0
[1461106196:332190035 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1461106196:332199425 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1461106196:332222300 OPENAVB osalTime] INFO: local_time = 1461088224168277063
[1461106196:332224629 OPENAVB osalTime] INFO: ml_phoffset = 0, ls_phoffset = 17972063438315
[1461106196:332234703 OPENAVB osalTime] INFO: ml_freqffset = 1.000000, ls_freqoffset = 0.999957
[1461106196:332240810 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.0 (Development)
[1461106196:332268117 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.0 (Development)
[1461106196:333116114 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1461106196:333124591 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1461106196:432777893 OPENAVB Endpoint] ERROR: Failed to initialize SRP
[1461106196:432784038 OPENAVB Endpoint] ERROR: Make sure that mrpd daemon is started.
```

Using GDB i've determined that when sending the "S??" command to the SRP Daemon, I receive "MSRP:Empty\n" in return instead of one of the other expected responses.

Does anyone know what would cause this?  Is something about my setup or the way I compiled incorrect?

intel-ethernet, 2016-04-30:
"MSRP:Empty” is returned by the daemon when there are no attributes at all (no streams registered – either by talkers or listeners).

I don’t know the avtp_pipeline well enough to say what conditions it fails with ‘Failed to initialize SRP’.

ekm

DMats, 2016-05-02:
Hello intel-ethernet.  Thanks for your prompt response.

**Analysis**
Using GDB I was able to compare the simple listener (and other examples) to the echo listener:  

In the simple listener, the `mrp_get_domain` call waits until `domain_a_valid` or `domain_b_valid` asserts high.  When I first start up the simple listener, it waits inside `mrp_get_domain` until I start up the simple talker on the other end.
```
while (!ctx->halt_tx && (ctx->domain_a_valid == 0) && (ctx->domain_b_valid == 0))
	usleep(20000);
```
In the echo listener, the `mrp_get_domain` call only sleeps 5 times before it returns failure, thus printing "Failed to initialize SRP" in the calling code.
```
int tries = 5;
while (!halt_tx && (domain_a_valid == 0) && (domain_b_valid == 0) && tries--) {
	usleep(20000);
```
Interestingly, after a day of playing around with the different examples with the same instances of the daemons running in the background, I attempted the simple listener for a second time. This time, the `mrp_get_domain` did not hang. It passed through the loop with `domain_b_valid == 1` immediately.  It continued until it reached the await_talker() call.

This leads me to believe that something else I did in the interim caused domain b to become valid, and the SRP daemon retained this state through until the second simple listener attempt.

Then I tried the echo_listener a second time as well.  Just like the simple_listener, it passed through the loop with `domain_b_valid == 1`.  However, the function still returned failure because the final line requires both domain_a and domain_b to be valid:
```
return domain_a_valid && domain_b_valid ? 0 : -1;
```

**Conclusion**
1. I still don't know why neither domain A nor domain B are valid at first start-up. That'll be investigation for tomorrow.
2. I believe last line `Open-AVB/lib/avtp_pipeline/openavb_common/mrp_client.c/mrp_get_domain()` should use OR (||) instead of AND (&&).

andrew-elder, 2016-05-03:
Agreed, the
  
 return domain_a_valid && domain_b_valid ? 0 : -1;
  
 line in mrp_get_domain() should contain an || operation.
  
  
  

----------------------------------------

DMats, 2016-05-04:
This issue seems to have yielded a few bugs worth fixing. I can try and make a pull request with the necessary changes.  I would submit the pull request to the open-avb-next branch correct?

## Simple Listener / Simple Talker Bug
I've determined why the simple talker/listener get stuck in their respective `mrp_get_domain` calls.  Actually, the issue had already been answered in issue #335 but the fix never made it into the code.  Like @marvinyane, I am also using an endpoint-to-endpoint topology rather than an endpoint-bridge-endpoint topology.  Thus, there is no bridge to make the initial domain attribute declaration (which would have let me pass through the `mrp_get_domain` call)

   To prevent others from getting stuck on this, I think `mrp_get_domain` in both [examples/common/talker_mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/examples/common/talker_mrp_client.c) and [examples/common/listener_mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/examples/common/listener_mrp_client.c)  should be changed to timeout after a certain number of tries, just like the [lib/avtp_pipeline/openavb_common/mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/openavb_common/mrp_client.c).  

## Echo Listener / Echo Talker (avtp_pipeline) Bug
After changing the final line of mrp_client.c `mrp_get_domain` to use an || operation rather than an && operation, it still returns failure because at start-up, neither domain a nor domain b are valid yet and there is no bridge to declare the initial domain attributes.  (ie. there are not yet any attributes registered with the SRP daemon).  In turn, this causes the entire SRP initialization to fail in both talker and listener endpoints before they can declare their own domain attributes.  `openavbSrpInitialize` exits before `mrp_register_domain` is called.

   [openavb_endpoint_osal.c](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/platform/Linux/endpoint/openavb_endpoint_osal.c) Line 274:
   ```c
	err = mrp_get_domain(&class_a_id,&a_priority, &a_vid, &class_b_id, &b_priority, &b_vid);
	if (err) {
		AVB_LOG_DEBUG("mrp_get_domain failed");
		goto error;
	}

	AVB_LOGF_INFO("detected domain Class A PRIO=%d VID=%04x...", a_priority, (int)a_vid);
	AVB_LOGF_INFO("detected domain Class B PRIO=%d VID=%04x...", b_priority, (int)b_vid);

	err = mrp_register_domain(&domain_class_a_id, &domain_class_a_priority, &domain_class_a_vid);
	if (err) {
		AVB_LOG_DEBUG("mrp_register_domain failed");
		goto error;
	}
```

   To fix this issue I believe that the `goto error;` line after `AVB_LOG_DEBUG("mrp_get_domain failed");` in the above code segment should be removed.  An endpoint should be allowed to declare domain attributes, even if it's the first AVB node to do so.

pinealservo, 2016-05-05:
I must have misread the discussion in #335; at the time I thought that the avtp_pipeline code was doing the right thing in its SRP interactions, but apparently it just borrowed the faulty logic from the examples directory. Thanks for bringing this up again!

Unfortunately, your suggested fix won't work correctly, because the various domain attribute variables are declared a few lines above without initialization. The call to `mrp_get_domain`, when successful, fills them in with the link peer's attribute values. When that call fails, they're still uninitialized, so there's only garbage in them to declare. The query and response parsing is actually going on asynchronously in another thread, which is why there's a delay

As per the discussion in the other issue, an endpoint should never wait for its link peer to declare the Domain attribute. It's okay to search to see if it's already declared one, but it's also okay to just immediately declare the default Domain attribute. When a Domain attribute from the link peer is registered (i.e. the link peer declared it and we received the MRP event regarding it) we need to check if it's the same as the Domain attribute we declared; if not, we must change our declaration to match the link peer. If you would like a detailed description of the SRP protocol, see http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf

The default Domain attribute depends on what traffic class you intend to support; Class A defaults would include SRClassID of 6, SRClassPriority of 3, and SRClassVID of 2.  Class B defaults are SRClassID 5, SRClassPriority 2, and SRClassVID of 2.

I'm not entirely sure what the best approach to take with the OpenAVB SRP implementation is. If I'm reading the MSRP code correctly, declaring a domain attribute will make the next query appear as if the neighbor has declared the same domain attribute, so checking first is probably best. I'll have to do some experiments with this. But if we query, wait a short while, then declare the defaults if no one has already declared domain attributes, we will probably get mostly-reasonable behavior. We should really wait for a matching Domain declaration before continuing, but I'm not seeing a way to do that right now.  I guess all attributes will be sent over the control connection when they're registered once you've issued a query, so maybe the SRP control socket message handler could watch for additional Domain attribute events after the first one due to the table dump and set a flag once one is seen. Maybe @andrew-elder or @intel-ethernet have some ideas?

Anyway, hopefully this helps get you on the right path.

andrew-elder, 2016-05-05:
Regarding the domain attribute, we implement something close to the query first, wait a short while and then declare if nothing else has declared. We keep track of what domain was declared and watch for incompatible domain declarations being reported by mrpd.

DMats, 2016-05-05:
@andrew-elder, I believe you are partially correct.  `openavbSrpInitialize` issues a query and waits a short while using `mrp_get_domain`.  Meanwhile, the `mrp_monitor_thread` is waiting for responses from the SRP daemon.   Upon hearing that no domain attributes are registered, `mrp_get_domain` returns failure, and `openavbSrpInitialize` exits without ever registering a new domain.  
Conversely, if the query actually returns a link peer's domain attributes, then they're stored to the global variables declared in [mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/openavb_common/mrp_client.c)
```c
volatile int domain_a_valid = 0;
int domain_class_a_id = 0;
int domain_class_a_priority = 0;
u_int16_t domain_class_a_vid = 0;

volatile int domain_b_valid = 0;
int domain_class_b_id = 0;
int domain_class_b_priority = 0;
u_int16_t domain_class_b_vid = 0;
```

I can't find the part of the code that watches for incompatible domain declarations reported by mrpd.

@pinealservo, you said:
> Unfortunately, your suggested fix won't work correctly, because the various domain attribute variables are declared a few lines above without initialization. The call to mrp_get_domain, when [...]

But, the local variables fed to `mrp_get_domain` are different from the global variables fed to `mrp_register_domain`, so it doesn't matter that they're left uninitialized upon failure. The follow-up `mrp_register_domain` calls are made using all zeros.

pinealservo, 2016-05-06:
> But, the local variables fed to mrp_get_domain are different from the global variables fed to mrp_register_domain, so it doesn't matter that they're left uninitialized upon failure. The follow-up mrp_register_domain calls are made using all zeros.

Regardless, that's not a valid domain declaration to make.  If you don't see a domain registration, you have to declare the default values I mentioned, not zeros.

> I can't find the part of the code that watches for incompatible domain declarations reported by mrpd.

He was probably referring to code in his company's usage of the OpenAVB SRP daemon, not anything in this repository.

andrew-elder, 2016-05-06:
Correct, I was referring to my company's application that is an mrpd client, not anything that is part of the OpenAVB repository.
  
  
    

I can't find the part of the code that watches for incompatible domain declarations reported by mrpd.   

He was probably referring to code in his company's usage of the OpenAVB SRP daemon, not anything in this repository.  

-

DMats, 2016-05-06:
Ohh!  I misunderstood what you were talking about.

I forked the repo and changed the logic to do what you said, query, wait, but then declare default values if no domains found.  I'll be able to test it on my setup next week.

CHM-me-ch, 2017-11-15:
@DMats 
Hi,
I am new to this and i am facing lot of issues while running most of the files for the first time.I tried running the simple talker  setup, but was facing 'PDelayResponse time' error and still not able to resolve this. The issue which i am pointing out now is:
i wanted to run the run_echo_talker.sh and for that, as you have mentioned above i run the following command $ ARCH=I210 make daemons_all after executing the 'make igb' command.

`username@username-desktop:~/Open-AVB$ ARCH=I210 make daemons_all
mkdir -p daemons/mrpd && make  subdir=daemons/mrpd  -C daemons/mrpd
make[1]: Entering directory '/home/username/Open-AVB/daemons/mrpd'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/username/Open-AVB/daemons/mrpd'
mkdir -p daemons/maap/linux/build/ && make  subdir=daemons/maap/linux/build/  -C daemons/maap/linux/build/
make[1]: Entering directory '/home/username/Open-AVB/daemons/maap/linux/build'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/username/Open-AVB/daemons/maap/linux/build'
mkdir -p daemons/gptp/linux/build/ && make  subdir=daemons/gptp/linux/build/  -C daemons/gptp/linux/build/
make[1]: Entering directory '/home/username/Open-AVB/daemons/gptp/linux/build'
g++ -Wall -g -I. -I../../common -I../src -I/home/username/header/include/ -DPTP_HW_CROSSTSTAMP -I../../../../lib/igb/ -DWITH_IGBLIB -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_i210.cpp -o obj/linux_hal_i210.o
In file included from ../src/linux_hal_common.hpp:48:0,
                 from ../src/linux_hal_generic.hpp:37,
                 from ../src/linux_hal_i210.cpp:34:
/usr/include/linux/ethtool.h:1222:58: error: ‘__KERNEL_DIV_ROUND_UP’ was not declared in this scope
  __u32 queue_mask[__KERNEL_DIV_ROUND_UP(MAX_NUM_QUEUE, 32)];
                                                          ^
Makefile:156: recipe for target 'obj/linux_hal_i210.o' failed
make[1]: *** [obj/linux_hal_i210.o] Error 1
make[1]: Leaving directory '/home/username/Open-AVB/daemons/gptp/linux/build'
Makefile:59: recipe for target 'gptp' failed
make: *** [gptp] Error 2
`

My setup is: Ubuntu 16.04 LTS machine with I210 card. Since I have only machine, i am trying to run only the talker side. Any help is highly appreciated. Thanks in advance

pinealservo, 2017-11-16:
@CHM-me-ch Running only the talker side doesn't work. The protocols are designed to ensure the requirements for defended stream reservations are met before streaming is attempted. The requirements include having at least two devices running the protocols.

If you just want to play around with scheduled transmit, you can start with simple_talker and just comment out the parts that try to talk with the mrp and gptp daemons, along with the bits that try to scale the timestamps with the information from the gptp daemon.

closed 2017-11-16 by pinealservo

CHM-me-ch, 2017-11-20:
okay, that was of so much help :) thanks a lot. I will work on this and post the result .Thanks again 👍