Avnu/OpenAvnu #383: Can I build Igb_avb driver for ARM platform?

ksrivastava11, 2016-05-03:
Hi,

I am trying to build the IGB_AVB driver for ARM platform.

Can you please guide me how I can build it. I am having the tool chain of ARM and kernel tree for the EVM for which I am building. 

I tried to build it but in Makefile, it expects that the KSP should be /lib/module but I am having a separate kernel tree. 

My build platform is Ubuntu 14.04
Kernel version of ubuntu is 3.13.0-32-generic
Kernel version for which I am building the driver is linux-3.0.35

pls can you help me..

Thanks,
Krishna

andrew-elder, 2016-05-03:
Krisha, building the IGB driver for ARM will be an R&D effort. I'm not aware that anyone has done it successfully yet. Good luck.

intel-ethernet, 2016-05-03:
The kernel loadable module *should* build on ARM platforms, although only others in our group have tried it with the standard igb driver from sourceforge. AFAIK though it should still compile, but perhaps with fixups.

The user-mode ‘igb’ library however would need some changes to remove dependencies on IA instructions (primarily ‘rdtsc’ and such).

ekm

ksrivastava11, 2016-05-03:
Hello ekm,

Well pointed, I was able to cross compile the driver for ARM, but I was unable to compile user-mode igb library. I am getting below error: 

igb.c: In function 'rdtscpll':
igb.c:1116:2: error: impossible constraint in 'asm'
make: *** [igb.o] Error 1

Now I need to find out the equivalent of the below assembly level code:  Any Suggestions or any guidance from where I can get some information ?

static inline void rdtscpll( uint64_t *val ) {
        uint32_t high, low;
        __asm__ __volatile__( "lfence;"
                                                  "rdtsc;"
                                                  : "=d"(high), "=a"(low)
                                                  :
                                                  : "memory" );
        *val = high;
        *val = (*val << 32) | low;
}

static inline void __sync() {
        __asm__ __volatile__( "mfence;"
                                                  :
                                                  :
                                                  : "memory" );
}

Thanks,
Krishna Srivastava

intel-ethernet, 2016-05-03:
I’m not an expert on ARM instructions, but I believe the nearest equivalents are ‘DSB’ for ‘lfence’ and ‘MRC p15, 0, r0, c15, c12, 1’ (where the results are stored in r0) for ‘rdtsc’. As I understand it, the performance counter is only available on some ARM cores.

ekm

ksrivastava11, 2016-05-04:
Hello ekm,

is it necessary to have DCA driver also with the igb_avb? do you have DCA source code, how I can download compile that for ARM? 

Thanks,
Krishna Srivastava

pinealservo, 2016-05-04:
I believe we're using the DCA driver that's distributed with the kernel as part of the Intel ethernet driver set. It's probably distributed in binary form with your target Linux distribution as a module, or you can build it again from the distribution's kernel source tree.

Regarding the use of the rdtsc instruction--ARM platforms vary quite a bit in the availability of timers, and you may want to investigate which timers your Linux distribution is using for high-resolution timing. You can probably do some of this investigation through the information that the clocksource subsystem exposes through sysfs, although you may need to dig into the SoC-specific support code in the kernel tree as well.

The purpose of the rdtscpll function in the IGB library is to cross-time-stamp the IGB's internal time counter with a high-resolution CPU-local time counter. This allows us to characterize the difference in clock rate between the two, and with that information we can directly translate and compare timestamps from one clock source to another. You'll want to use the highest-resolution timer you have available to you in order to keep the time-related error low. 

We're very happy to answer any questions related to AVB or the code as it exists right now, but this is not a general-purpose support forum for Linux issues or cross-platform porting efforts. Most of us do our support here on our own time or as a a small part of our responsibilities to our employers, so please be mindful of this as you look for help in your efforts. We'll be able to help you much more efficiently if you keep questions here very focused on specifics of the current code base rather than asking for general guidance in doing something that's not currently supported.

intel-ethernet, 2016-05-04:
You shouldn’t need the DCA module – its an Intel-platform specific feature. You should be able to compile it out by either changes to the Makefile or command-line ‘undefine’.

ekm

ksrivastava11, 2016-05-19:
Thanks ekm for your support, I got the answers I wanted.

closed 2016-05-19 by andrew-elder