Avnu/OpenAvnu #384: top level make failure

DMats, 2016-05-04:
Currently, if you attempt to `make all` from the top level on the master branch, it will fail due to errors in the jackd-talker example:
```
[labuser@localhost Open-AVB-master]$ make clean all
mkdir -p kmod/igb/ && make  subdir=kmod/igb/  -C kmod/igb/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb'
rm -f -r igb_avb.ko igb_avb.o igb_avb.mod.c igb_avb.mod.o igb_main.o e1000_82575.o e1000_i210.o e1000_mac.o e1000_nvm.o e1000_phy.o e1000_manage.o igb_param.o igb_ethtool.o kcompat.o e1000_api.o e1000_mbx.o igb_vmdq.o igb_procfs.o igb_hwmon.o igb_debugfs.o igb_ptp.o \
igb_avb.7.gz .*cmd .tmp_versions Module.markers Module.symvers modules.order
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb'
mkdir -p lib/igb/ && make  subdir=lib/igb/  -C lib/igb/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/lib/igb'
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/lib/igb'
mkdir -p daemons/mrpd/ && make  subdir=daemons/mrpd/  -C daemons/mrpd/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/daemons/mrpd'
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
rm -f mrpd mrpctl
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/daemons/mrpd'
mkdir -p daemons/gptp/linux/build/ && make  subdir=daemons/gptp/linux/build/  -C daemons/gptp/linux/build/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/daemons/gptp/linux/build'
rm -f *~ obj/*.o  obj/daemon_cl 
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/daemons/gptp/linux/build'
mkdir -p daemons/maap/linux/ && make  subdir=daemons/maap/linux/  -C daemons/maap/linux/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/daemons/maap/linux'
rm -rf ./maap_daemon
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/daemons/maap/linux'
mkdir -p examples/simple_talker/ && make  subdir=examples/simple_talker/  -C examples/simple_talker/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/simple_talker'
rm -f simple_talker
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/simple_talker'
mkdir -p examples/simple_listener/ && make  subdir=examples/simple_listener/  -C examples/simple_listener/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/simple_listener'
rm -f  simple_listener
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/simple_listener'
mkdir -p examples/mrp_client/ && make  subdir=examples/mrp_client/  -C examples/mrp_client/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/mrp_client'
rm -f mrpl mrpq mrpValidate
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/mrp_client'
mkdir -p examples/jackd-talker/ && make  subdir=examples/jackd-talker/  -C examples/jackd-talker/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/jackd-talker'
rm -f jackd_talker
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/jackd-talker'
mkdir -p examples/jackd-listener/ && make  subdir=examples/jackd-listener/  -C examples/jackd-listener/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/jackd-listener'
rm -f jack_listener
rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"`
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/jackd-listener'
mkdir -p examples/live_stream/ && make  subdir=examples/live_stream/  -C examples/live_stream/ clean
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/live_stream'
rm -f talker listener
rm -f *.o *~ *.rej *.orig
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/live_stream'
make -s -C lib/avtp_pipeline -f avtp_pipeline.mk clean
mkdir -p kmod/igb && make  subdir=kmod/igb  -C kmod/igb 
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb'
make -C /lib/modules/3.10.0-327.el7.x86_64/build SUBDIRS=/home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb modules
make[2]: Entering directory `/usr/src/kernels/3.10.0-327.el7.x86_64'
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/igb_main.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/e1000_82575.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/e1000_i210.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/e1000_mac.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/e1000_nvm.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/e1000_phy.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/e1000_manage.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/igb_param.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/igb_ethtool.o
  CC [M]  /home/labuser/Desktop/AVB_orig/Open-AVB-master/kmod/igb/kcompat.o
[... 99 lines cut ...]
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/live_stream'
cc -O2 -g -Wall -Wextra -Wno-parentheses -I ../../lib/igb/ -I../../daemons/mrpd -I../common -I../../daemons/common  -c talker.c
cc -L ../../lib/igb/ talker.o ../common/avb.o ../common/talker_mrp_client.o -ligb -lpci -lrt -lpthread  -o talker
cc -O2 -g -Wall -Wextra -Wno-parentheses -I ../../lib/igb/ -I../../daemons/mrpd -I../common -I../../daemons/common  -c listener.c
cc -L ../../lib/igb/ listener.o ../common/avb.o ../common/listener_mrp_client.o -ligb -lpci -lrt -lpthread  -o listener
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/live_stream'
make lib
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master'
mkdir -p lib/igb && make  subdir=lib/igb  -C lib/igb 
make[2]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/lib/igb'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/lib/igb'
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master'
mkdir -p examples/jackd-talker && make  subdir=examples/jackd-talker  -C examples/jackd-talker 
make[1]: Entering directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/jackd-talker'
cc -c -I../../lib/igb -I../../daemons/mrpd -I../common -I../../daemons/common -I../../daemons/mrpd -O2 -g -I../../lib/igb -I../../daemons/mrpd -I../common -I../../daemons/common -Wall -Wextra -Wno-parentheses -std=gnu99 jackd_talker.c
jackd_talker.c:48:23: fatal error: jack/jack.h: No such file or directory
 #include <jack/jack.h>
                       ^
compilation terminated.
make[1]: *** [jackd_talker.o] Error 1
make[1]: Leaving directory `/home/labuser/Desktop/AVB_orig/Open-AVB-master/examples/jackd-talker'
make: *** [jackd-talker] Error 2
```
I wasn't sure how to fix this, so I merely hacked the top level Makefile to skip jackd_talker and jackd_listener when `make all` is called.  Is anyone familiar with the jackd_talker example who knows a better solution?

pinealservo, 2016-05-04:
You need the development package for the jack library installed in order to build that one. To see how the CI environment is set up, check out the `.travis.yml` and `travis.sh` files at the top level of the repository.  Some of those commands (especially the kernel-related ones in the YAML file) shouldn't be necessary on newer systems than the ones running on Travis, but it should give you a good idea of what you can expect to build.

Relevant to your current problem is the `libjack-dev` package, if you're using a Ubuntu or other Debian-derived system.

DMats, 2016-05-05:
Hey pinealservo, 
Thank you for the prompt response.

I'm running on CentOS7.  Unfortunately, my attempts to use yum with the packages mentioned in `.travis.yml` failed.  
```
sudo yum install libpcap-dev libpci-dev libsndfile1-dev libjack-dev linux-headers-3.13.0-36-generic cmake
sudo yum install libgstreamer0.10-dev libgstreamer-plugins-base0.10-dev libasound2-dev
sudo yum search libjack
sudo yum search jack
```

pinealservo, 2016-05-05:
Apparently the CentOS package name for the development package is `jack-audio-connection-kit-devel`. I'm not very familiar with CentOS, though, so I don't know if it's in the default repository that yum searches or if you'll have to get the RPM elsewhere.  I found it with a search on rpm.pbone.net.

andrew-elder, 2016-05-05:
I wonder if we need to add a packages "check" section to the top level makefile?

DMats, 2016-05-05:
Ahah!  my top level `make all` works now that I've installed the jack-audio-connection-kit-devel dependency.  Thank you @pinealservo.

I downloaded two rpms from rpm.pbone.net:
1. [jack-audio-connection-kit](http://rpm.pbone.net/index.php3/stat/4/idpl/30480018/dir/centos_7/com/jack-audio-connection-kit-0.121.3-2.1.x86_64.rpm.html)
2. [jack-audio-connection-kit-devel](http://rpm.pbone.net/index.php3/stat/4/idpl/30480020/dir/centos_7/com/jack-audio-connection-kit-devel-0.121.3-2.1.x86_64.rpm.html)

And installed them like so:
```
[labuser@localhost Desktop]$ sudo rpm -ivh ../Downloads/Packages/jack-audio-connection-kit-0.121.3-2.1.x86_64.rpm 
warning: ../Downloads/Packages/jack-audio-connection-kit-0.121.3-2.1.x86_64.rpm: Header V3 RSA/SHA1 Signature, key ID 427beafd: NOKEY
Preparing...                          ################################# [100%]
Updating / installing...
   1:jack-audio-connection-kit-0.121.3################################# [100%]
[labuser@localhost Desktop]$ sudo rpm -ivh ../Downloads/Packages/jack-audio-connection-kit-devel-0.121.3-2.1.x86_64.rpm 
warning: ../Downloads/Packages/jack-audio-connection-kit-devel-0.121.3-2.1.x86_64.rpm: Header V3 RSA/SHA1 Signature, key ID 427beafd: NOKEY
Preparing...                          ################################# [100%]
Updating / installing...
   1:jack-audio-connection-kit-devel-0################################# [100%]
```

@andrew-elder Would the packages check reside in the top level make file? or should it be part of the lowest level makefile that the dependency is required for?  Or both?  sounds like a good idea, but I have no idea how to implement one.

pinealservo, 2016-05-24:
There's a tool called pkg-config (http://pkg-config.freedesktop.org) that can be used by build systems to query the presence and useful information about libraries that support it.  It turns out that jack (at least as packaged by Ubuntu) provides the necessary .pc file to query the presence of jack.  There may be some way to get Cmake to ask pkg-config if jack is installed; it looks like the [FindPkgConfig]:https://cmake.org/cmake/help/v3.0/module/FindPkgConfig.html command is supposed to do it.