Avnu/OpenAvnu #385: Error in avtp_pipeline/README.md

DMats, 2016-05-04:
**Background**
I'm building from the repo root on an Intel PC running Linux operating system with Intel I210 NIC card installed.  

**Issue**
There is a slight error in the [avtp_pipeline/README.md](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/README.md)
```
    Building from the repo root
    $ ARCH=I210 make all
```
The above code snippet fails because both the igb_avb kernel driver and the gptp daemon Makefiles use the ARCH variable.

The code snippet shown below, taken from the [igb_avb driver Makefile](https://github.com/AVnu/Open-AVB/blob/master/kmod/igb/Makefile) shows that the ARCH variable is assigned based on the architecture it's being built on.
```make
# we need to know what platform the driver is being built on
# some additional features are only built on Intel platforms
ifeq (,$(ARCH))
 ARCH := $(shell uname -m | sed 's/i.86/i386/')
endif
ifeq ($(ARCH),alpha)
  EXTRA_CFLAGS += -ffixed-8 -mno-fp-regs
endif
ifeq ($(ARCH),x86_64)
  EXTRA_CFLAGS += -mcmodel=kernel -mno-red-zone
endif
ifeq ($(ARCH),ppc)
  EXTRA_CFLAGS += -msoft-float
endif
ifeq ($(ARCH),ppc64)
  EXTRA_CFLAGS += -m64 -msoft-float
  LDFLAGS += -melf64ppc
endif
```

The code snippet shown below, taken from the [gPTP daemon Makefile](https://github.com/AVnu/Open-AVB/blob/master/daemons/gptp/linux/build/Makefile) shows that the ARCH must equal I210 to compile for the Intel I210 NIC card.
```make
ifeq ($(ARCH),I210)
	IGB_LIB_INCPATH=../../../../lib/igb/
	IGB_LIB_PATH=../../../../lib/igb/
	CFLAGS_G += -I$(IGB_LIB_INCPATH) -DWITH_IGBLIB
	LDFLAGS_G += -lz -ligb -lpci -L$(IGB_LIB_PATH)
	OBJ_FILES += $(OBJ_DIR)/linux_hal_generic.o \
		$(OBJ_DIR)/linux_hal_generic_adj.o
	OBJ_FILES += $(OBJ_DIR)/linux_hal_i210.o 
	HEADER_FILES += $(SRC_DIR)/linux_hal_generic.hpp
else ifeq ($(ARCH),IntelCE)
	INTELCE_INCPATH=/home/hitesh/work/smd/Dinerout/i686-linux-elf/include/
	INTELCE_LINUX_INCPATH=/home/hitesh/work/smd/Dinerout/i686-linux-elf/include/linux_user
	INTELCE_LIBPATH=/home/hitesh/work/smd/Dinerout/i686-linux-elf/lib/
	CFLAGS_G += -I$(INTELCE_LINUX_INCPATH) -I$(INTELCE_INCPATH) -DARCH_INTELCE
	LDFLAGS_G += -L$(INTELCE_LIBPATH) -lismd_clock_ioctl -lismd_core_ioctl
	HEADER_FILES += $(SRC_DIR)/linux_hal_intelce.hpp
	OBJ_FILES += $(OBJ_DIR)/linux_hal_intelce.o
	CXX = $(TARGETCXX)
else
	OBJ_FILES += $(OBJ_DIR)/linux_hal_generic.o \
		$(OBJ_DIR)/linux_hal_generic_adj.o
	HEADER_FILES += $(SRC_DIR)/linux_hal_generic.hpp
endif
```

**Short-term Solution**
Until this shared dependency is resolved, I have been compiling at the top level using commands very similar to those found in [travis.sh](https://github.com/AVnu/Open-AVB/blob/master/travis.sh):
```
make igb
make lib
ARCH=I210 make daemons_all
make examples_all
make avtp_pipeline
```

andrew-elder, 2016-05-05:
It certainly seems a little strange to have I210 be an ARCH setting. Thanks for bringing it up.

pinealservo, 2016-05-24:
Further discussion regarding the build problem can be found in #390; linking these so we remember to resolve this documentation issue when the build is resolved, since the resolution to #390 is likely to change the instructions to some degree.