Avnu/OpenAvnu #388: gPTP - Neighbor Prop Delay Threshold Issues

pinealservo, 2016-05-05:
I came across some issues while working on the Automotive Profile branch.  I'm getting it rebased on the latest open-avb-next, since quite a lot of development has occurred since it was first implemented.

1. There's no debug output when ASCapable changes due to PDelay exceeding Neighbor Prop Delay Threshold, so it's not clear why ASCapable is being disabled.

2. It's often useful to be able to change the Neighbor Prop Delay to deal with non-ideal situations while testing (network taps, media converters, etc.) but it's currently just a #defined constant.

3. The code that sets ASCapable in `PTPMessagePathDelayRespFollowUp::processMessage` currently sets it to true when it can make a valid calculation, but then sets it false again if it's beyond the Neighbor Prop Delay Threshold. This leads to a pair of log messages per PDelay transaction when the threshold is exceeded.

These aren't really critical things, and I'll be fixing them soon, but I thought I'd put the issue up for discussion in the mean time.

andrew-elder, 2016-05-05:
@pinealservo - I was wondering what you think about creating a temporary "merging-auto" branch or some such thing. That way others can review the merging work and we can maintain merging-auto for some amount of time in AVnu repos.

kencarlino, 2016-05-05:
In the Automotive Profile asCapable is set to True and should stay True. The disabling of asCapable state changes should only happen when the Automotive Profile is enabled. 

See 9.3.4 neighborPropDelay value in the AutoCDS spec (1.4) in case more clarification is needed.

pinealservo, 2016-05-24:
I believe this is now resolved in open-ave-next since #405 was merged. It contains the following resolutions:

1. Adds debug output when the neighbor propagation delay exceeds the threshold.
2. Does not make asCapable false due to propagation delay when the automotive profile is enabled.
3. Allows overriding the default allowed propagation delay via the INI file in Linux.

closed 2016-05-24 by pinealservo