Avnu/OpenAvnu #389: gPTP Message Logging

pinealservo, 2016-05-05:
When the automotive profile work was started, debug output was still printed via raw printfs. The authors of that work created a multi-level log interface and re-targeted all the debug output to the new interface. Meanwhile, the printfs were replaced in the main development line with a different log/debug interface.

My merge of the two still has both log interfaces, but we should probably consolidate to one of them.

The `open-avb-next` interface is here: https://github.com/AVnu/Open-AVB/blob/open-avb-next/daemons/gptp/linux/src/debugout.hpp and https://github.com/AVnu/Open-AVB/blob/open-avb-next/daemons/gptp/windows/daemon_cl/debugout.hpp

The `feature-gptp-avnu-automotive-profile` interface is here: https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/gptp_log.cpp and https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/gptp_log.hpp

Does anyone have any strong feelings about which to use or how to combine them?  I like the formatting and timestamping of the gptpLog function, but it probably needs to be made arch-specific like the current debug print macros are.

andrew-elder, 2016-05-05:
gptpLog() is much more flexible and capable, so to my way of thinking that is the to keep. It may cause some temporary hiccups in the Windoze build, but I'm sure that can be worked out.

christopher-s-hall, 2016-05-05:
I would ask that we not break the Windows build. Intel has customers who depend on it and I don’t have the “spare time” to fix the build issues introduced by others. I test all of my changes on both OSs and don’t see any reason why others can’t do the same.

This is comment doesn’t specifically address gptpLog() which may be fine.

Chris

andrew-elder, 2016-05-06:
@christopher-s-hall - I assume you meant "please don't break the Windows build in open-avb-next"? Or, did you mean "please don't break the Windows build in "master". I assume you don't object to the Windows build being broken on some new (temporary) merging branch?

pinealservo, 2016-05-12:
I believe I resolved any issues that gptpLog would have caused with the Windows build, and didn't even need to move the gptp_log.cpp to the platform-specific directories.  I'll get the logging calls moved over.  Thanks for the feedback.

closed 2016-05-12 by pinealservo