Avnu/OpenAvnu #390: Unable to build gPTP on open-avb-next branch

DMats, 2016-05-05:
**Background**
I'm on an x86_64 PC with CentOS7 OS and 2 I210 NIC cards.

**Problem**
I'm unable to build gPTP on the open-avb-next branch.  I think I need to use cmake but i'm not sure how.

**Attempt 1**, build from top level:
```
[labuser@localhost Open-AVB-open-avb-next]$ ARCH=I210 make daemons_all
mkdir -p daemons/mrpd && make  subdir=daemons/mrpd  -C daemons/mrpd 
make[1]: Entering directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/daemons/mrpd'
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o mrpd.o  mrpd.c
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o mvrp.o  mvrp.c
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o msrp.o  msrp.c
msrp.c: In function ‘msrp_count_type’:
msrp.c:135:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (attrib_type == attrib->type) {
                    ^
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o mmrp.o  mmrp.c
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o mrp.o  mrp.c
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o parse.o  ../common/parse.c
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o eui64set.o  ../common/eui64set.c
cc  mrpd.o mvrp.o msrp.o mmrp.o mrp.o parse.o eui64set.o  -o mrpd
cc -c -I../common -I../../examples/mrp_client -O2 -Wall -Wextra -Wno-parentheses -ggdb -D_GNU_SOURCE -o mrpctl.o  mrpctl.c
make -C ../../examples/mrp_client/ mrpdclient.o
make[2]: Entering directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/examples/mrp_client'
cc -c  -I../../daemons/mrpd -O2 -Wall -Wextra -Wno-parentheses -ggdb mrpdclient.c
make[2]: Leaving directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/examples/mrp_client'
cc  mrpctl.o ../../examples/mrp_client/mrpdclient.o  -o mrpctl
make[1]: Leaving directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/daemons/mrpd'
mkdir -p daemons/maap/linux/ && make  subdir=daemons/maap/linux/  -C daemons/maap/linux/ 
make[1]: Entering directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/daemons/maap/linux'
cc -o ./maap_daemon ../common/maap_protocol.c  maap_linux.c  -Wall -lpthread -lpcap -I../common 
make[1]: Leaving directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/daemons/maap/linux'
mkdir -p daemons/gptp/linux/build/ && make  subdir=daemons/gptp/linux/build/  -C daemons/gptp/linux/build/ 
make[1]: Entering directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/daemons/gptp/linux/build'
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../../common/ptp_message.cpp -o obj/ptp_message.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../../common/avbts_osnet.cpp -o obj/avbts_osnet.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../../common/ieee1588port.cpp -o obj/ieee1588port.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../../common/ieee1588clock.cpp -o obj/ieee1588clock.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../src/linux_hal_common.cpp -o obj/linux_hal_common.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../src/platform.cpp -o obj/platform.o
cc -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -I../../../../lib/igb/ -DWITH_IGBLIB -c ../../common/ini.c -o obj/ini.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../../common/gptp_cfg.cpp -o obj/gptp_cfg.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../src/linux_hal_generic.cpp -o obj/linux_hal_generic.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../src/linux_hal_generic_adj.cpp -o obj/linux_hal_generic_adj.o
g++ -Wall -g -I. -I../../common -I../src -I/home/labuser/header/include/ -Wnon-virtual-dtor  -c ../src/linux_hal_i210.cpp -o obj/linux_hal_i210.o
../src/linux_hal_i210.cpp:40:17: fatal error: igb.h: No such file or directory
 #include <igb.h>
                 ^
compilation terminated.
make[1]: *** [obj/linux_hal_i210.o] Error 1
make[1]: Leaving directory `/home/labuser/Desktop/DMats/Open-AVB-open-avb-next/daemons/gptp/linux/build'
make: *** [gptp] Error 2
```

**Attempt 2**, cmake:
```
[labuser@localhost gptp]$ cmake -G "Unix Makefiles"
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:10 (target_link_libraries):
  Cannot specify link libraries for target "gptp" which is not built by this
  project.

-- Configuring incomplete, errors occurred!
```

Does anyone know what i'm doing wrong?

pinealservo, 2016-05-06:
Doing `make daemons_all` should do it, as should `make gptp`. But it relies on the IGB library, so you need to `make lib` first.

DMats, 2016-05-09:
Thanks for the prompt response.  

You are right that both `make daemons_all` and `make gptp` complete successfully.  I made sure to run `make lib` first.  

For me, gptp fails to build on the open-avb-next branch when I set the variable ARCH=I210.  

For giggles, I also tried ARCH=IntelCE and it also built successfully.

hrtmnn, 2016-05-24:
Hey DMats,

i ran into the same issue as you did. I am on Ubuntu 14.04 64 Bit and have not been able to build gPTP with ARCH=I210 on open-avb-next branch due to missing igb.h file. Master branch did not work as well. 

I am not sure about the changes but i had success building the gPTP on feature-gptp-avnu-automotive-profile branch.

andrew-elder, 2016-05-24:
feature-gptp-avnu-automotive-profile was merged to open-avb-next over the weekend. I'd be interested to learn whether open-avb-next now works for you.

DMats, 2016-05-24:
Since I created this issue, the `open-avb-next` branch was merged into the master branch.  Perhaps that could explain why the master branch isn't working for you now.  I'm not sure why it thinks that `igb.h` is missing.  The file is still part of the code base.

hrtmnn, 2016-05-24:
I just did a rebuild and the issue still remains on open-avb-next.

pinealservo, 2016-05-24:
I think this has to do with a fairly recent merge of Intel stuff to open-avb-next in which the meaning of ARCH to some makefile (the IGB lib, I think?) was changed to mean the platform CPU architecture; having it set to I210 screws up some new logic that defaults it to x86 when not specified. Ultimately, one of the usages of ARCH needs to change to another name in order to allow building everything with an environment variable set to the architecture of the network interface.

andrew-elder, 2016-09-22:
@Dmats - can I assume this issue has now been addressed and you no longer have and issue?

DMats, 2016-09-22:
Yes this is fixed.  It was a makefile change, missing colon or something
like that.  I can't remember which pull request.

closed 2016-09-22 by andrew-elder