Avnu/OpenAvnu #391: Reloading igb_avb.ko crashes the OS

RikusW, 2016-05-06:
Doing insmod rmmod and then insmod again will crash the system entirely requiring a reboot.
I used kernel 3.19.0-15-generic #15-Ubuntu SMP 64 bit.

DMats, 2016-05-09:
I'm working with two intel x86_64 PCs running CentOS7 with two intel I210 cards installed, and I don't run into this issue.  When I run the `run_igb.sh` script with an added line to rmmod igb_avb, the PC does not crash.
```bash
rmmod igb
rmmod igb_avb # this is the line I added
modprobe i2c_algo_bit
modprobe dca
modprobe ptp
insmod kmod/igb/igb_avb.ko
```

eric-mann, 2016-05-09:
Not much meat to chew on with this bug report. Backtrace? 

You can get similar bad behavior if  you fail to unload the igb driver first. Two drivers operating on the same hw at the same time has 'undetermined behavior'.

RikusW, 2016-05-10:
It only happened when I did "rmmod igb" first, when I left it in there was no crash.
I'll try again with the above script.

RikusW, 2016-05-13:
For now I'll just avoid doing rmmod, then all is fine or so it seems.
I''m working for Uman an AVnu member company and will be working with Open AVB.

RikusW, 2016-05-13:
This is the required commands to make it crash, if igb is left in there is no crash.

rmmod igb
insmod kmod/igb/igb_avb.ko
rmmod igb_avb
insmod kmod/igb/igb_avb.ko

Totally dead after the last command.

eric-mann, 2016-05-13:
You definitely *do not* want to leave igb loaded.

Can you define ‘totally dead’?

ekm

RikusW, 2016-05-13:
Ironically leaving in igb prevents the crash, maybe because it prevents allocation of some resource.

The mouse and keyboard is unresponsive, numlock doesn't even change the led.
I had to unplug the PC to reboot since the reset button is not connected.
I haven't tried sysrq since I don't think its enabled.

eric-mann, 2016-05-13:
So is igb_avb already loaded when you try to load it again?

Can you include the output of ‘lsmod’ before you run the script?

ekm

RikusW, 2016-05-14:
Only igb is loaded beforehand. Removing igb_avb and then reloading it crashes the PC only if igb is NOT loaded. I'll do lsmod when I'm back at work.

RikusW, 2016-05-16:
$ lsmod
Module                  Size  Used by
pci_stub               16384  1
vboxpci                24576  0
vboxnetadp             28672  0
vboxnetflt             28672  0
vboxdrv               425984  3 vboxnetadp,vboxnetflt,vboxpci
cfg80211              548864  0
binfmt_misc            20480  1
snd_hda_codec_realtek    86016  1
snd_hda_codec_generic    69632  1 snd_hda_codec_realtek
snd_hda_intel          36864  3
snd_hda_controller     32768  1 snd_hda_intel
snd_hda_codec         143360  4 snd_hda_codec_realtek,snd_hda_codec_generic,snd_hda_intel,snd_hda_controller
snd_usb_audio         184320  2
gpio_ich               16384  0
snd_usbmidi_lib        36864  1 snd_usb_audio
snd_hwdep              20480  2 snd_usb_audio,snd_hda_codec
intel_powerclamp       20480  0
snd_pcm               106496  4 snd_usb_audio,snd_hda_codec,snd_hda_intel,snd_hda_controller
snd_seq_midi           16384  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            32768  2 snd_usbmidi_lib,snd_seq_midi
coretemp               16384  0
i915                 1060864  4
kvm_intel             151552  0
kvm                   487424  1 kvm_intel
snd_seq                65536  2 snd_seq_midi_event,snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_rawmidi,snd_seq_midi
serio_raw              16384  0
mei_me                 20480  0
video                  20480  1 i915
snd_timer              32768  2 snd_pcm,snd_seq
mei                    90112  1 mei_me
drm_kms_helper        131072  1 i915
snd                    90112  22 snd_hda_codec_realtek,snd_usb_audio,snd_hwdep,snd_timer,snd_pcm,snd_seq,snd_rawmidi,snd_hda_codec_generic,snd_usbmidi_lib,snd_hda_codec,snd_hda_intel,snd_seq_device
drm                   348160  6 i915,drm_kms_helper
lpc_ich                24576  0
shpchp                 40960  0
soundcore              16384  2 snd,snd_hda_codec
mac_hid                16384  0
parport_pc             32768  0
ppdev                  20480  0
lp                     20480  0
parport                45056  3 lp,ppdev,parport_pc
autofs4                40960  2
hid_generic            16384  0
usbhid                 53248  0
hid                   110592  2 hid_generic,usbhid
ahci                   36864  2
libahci                32768  1 ahci
igb                   188416  0
i2c_algo_bit           16384  2 igb,i915
r8169                  81920  0
dca                    16384  1 igb
ptp                    20480  1 igb
mii                    16384  1 r8169
pps_core               20480  1 ptp