Avnu/OpenAvnu #392: Getting AVB to run on Intel 82579V

Tigerclaw1993, 2016-05-06:
Hi everyone!

For my bachelor thesis I am trying to get the Open-AVB master branch running on a laptop (Toshiba Portege R830-112) using an Intel 82579V networkcard with OpenSuse 13.2 as its operating system. The networkcard is using the Intel e1000e driver instead of the (AVB-)igb driver.
So one chapter of my work will be to customize the 82579V / Intel e1000e driver, making it AVB capable.

The Intel 82579V ([Datasheet](http://www.intel.com/content/www/us/en/embedded/products/networking/82579-gbe-phy-datasheet-vol-2-1.html)) supports 802.1as for time synchronization and 802.1Q for stream reservation. So it should provide everything I need for the AVB implementation.

Additionally to the laptop, I got two ["AVB Audio Endpoint" boards](https://www.xmos.com/download/private/AVB-Audio-Endpoint-Product-Brief%281.1%29.pdf) from XMOS which are working really well with each other. These boards were bought to test the functionality of the resulting software. I would like to use them in a Talker (AVB board) to Listener (laptop) and Talker (laptop) to Listener (AVB board) test setup.

With the help of Wireshark I already found out that the gPTP Daemon is able to communicate with the XMOS board.

After recompiling the Linux Kernel to enable ptp these are my first steps (everything is done as superuser):
- Connecting laptop with the running XMOS board which is running as talker
- Running the gPTP daemon
`groupadd ptp`
`.../Open-AVB/daemons/gptp/linux/build/obj> ./daemon_cl enp0s25`
> Using clock device: /dev/ptp0
> Starting PDelay
> *** Announce Timeout Expired - Becoming Master
> New Grandmaster "E8:9D:87:FF:FE:AE:58:22" (previous "00:00:00:00:00:00:00:00")
> AsCapable: Enabled
> AsCapable: Disabled
> AsCapable: Enabled
> AsCapable: Disabled
...

- Start Stream, V-lan and MAC reservation
`.../Open-AVB/daemons/mrpd> ./mrpd -mvs -i enp0s25`
> process_events()

- Start the simple_listener
`.../Open-AVB/examples/simple_listener> ./simple_listener -i enp0s25 -f out.wav`
> Msg: MSRP:Empty

then stucking in a while loop.

According to the outprint of the simple_listener "Msg: MSRP:EMPTY" there is no registered stream reservation. So why does this fail? Unfortunately there is no error occurring when running mrpd.
Is this already the case of not being able to access the right adresses via the nic driver? Or am I missing something?

eric-mann, 2016-05-09:
Hmmm ... 

Well, here's what I think could work:
a) the gPTP daemon should 'just work' out of the box with the existing drivers
b) the mrpd daemon should also 'just work' out of the box with existing drivers
c) ... and running 'simple_listener' should also just work - as you don't even really need gPTP to receive a stream and record the samples to a .wav file. 

If you wanted to get fancy you would synchronize the PTP clocks and then discipline the audio clock (e.g. try to match frequency) to the clock implicit in the presentation times in the 1722 AVB (audio) packets. 

Doing a 'simple_talker' with the Intel 82579V would be more challenging - basically you would need to write a pure software-based credit based shaper, which I know is possible from prior experience, but usually either consumes "a fair number" CPU cycles to schedule and transmit a lot of streams, or it doesn't fully utilize the allocated bandwidth. 

As to the XMOS (and others can correct me if I'm wrong) - I though you needed what's called a controller (a 1722.1 AVDECC entity) to configure and start the talker stream on the XMOS box. You also need to configure simple_listener to "listen" for the matching stream ID as created on the talker.

Hope this helps - good luck.

DMats, 2016-05-09:
I ran into the same issue when trying out the simple_listener example, because similarly, I'm working with only 2 AVB endpoints without an AVB bridge inbetween.  

The loop it's stuck in can be found on Line 261 of `Open-AVB/examples/common/listener_mrp_client.c' inside the mrp_get_domain() function.
```
while (!ctx->halt_tx && (ctx->domain_a_valid == 0) && (ctx->domain_b_valid == 0))
		usleep(20000);
```

In the code snippet below, `mrp_get_domain` is called before `report_domain_status`.  What's happening is it's trying to query for existing AVB domain attributes that would have been provided by other link peers.  But since none yet exist, it gets stuck in the mrp_get_domain call, and never gets to the `report_domain_status` call at which point it would declare it's own AVB domain attributes.

From simple_listener.c main():
```
    rc=mrp_get_domain(ctx, class_a, class_b);
	if (rc)
	{
		printf("failed calling mrp_get_domain()\n");
		return EXIT_FAILURE;
	}

	printf("detected domain Class A PRIO=%d VID=%04x...\n",class_a->priority,class_a->vid);

	rc = report_domain_status(class_a,ctx);
	if (rc) {
		printf("report_domain_status failed\n");
		return EXIT_FAILURE;
	}
```

Check out [lib/avtp_pipeline/openavb_common/mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/openavb_common/mrp_client.c).  It was based heavily on the listener_mrp_client.c and talker_mrp_client.c. `mrp_get_domain` exists there as well, except the while loop in question times out after 5 iterations, a change that the simple_listener example would benefit from as well.

Tigerclaw1993, 2016-05-10:
Tanks for your replies!

In response to eric-mann:
I would like to start with an easy listener and will have a look at it again afterwards.
A running simple talker would be great for the thesis especially when I had to do some parts of it on my own, lets see how far I get ;)

To DMats:
I had a look at the code to see where the program is running into this while-loop as well. At this time I didn't know what to do with this information. For me, 'get_domain()' was not that meaningful.
But you're saying we do not really need this domain, don't you?

I now modified the loop to exit after some iterations, like it's done in the mrp_client.c and will try to get a established connection about the virtual LAN.
In my Wireshark captures I found out the XMOS Board is building up its connection on VLAN-ID 0001 and adapted the VLan-ID of the simple_listener to it. 

You may want to help me with these outputs:
> Msg: MSRP:Empty
> 
> detected domain Class A PRIO=0 VID=0001...
> Joing VLAN V++:I=0001
> 
> Waiting for talker...
> Msg: SJO D:C=0,P=0,V=0001,N=0 R=000000000000 VP/IN
> 
> Msg: VNE 0001 R=000000000000 VN/MT
> 
> Msg: SLE D:C=0,P=0,V=0001,N=0 R=000000000000 QA/MT

What are the meanings of?
- SJO (S = stream, JO = joined ?)
- VNE (V = virtual LAN?)
- SLE

Regards,
Chris

DMats, 2016-05-10:
@Tigerclaw1993 
I believe those "Msg: ..." lines are being printed at [examples/common/listener_mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/examples/common/listener_mrp_client.c) Line 87

I'll be honest, I don't fully understand the messages yet, but I think the following comment explains it.
[daemons/mrpd/mrpd.c](https://github.com/AVnu/Open-AVB/blob/master/daemons/mrpd/mrpd.c) Line 275:
```c
	/*
	 * Inbound/output commands from/to a client:
	 *
	 * When sent from a client, indicates an operation on the
	 * internal attribute databases. When sent by the daemon to
	 * a client, indicates an event such as a new attribute arrival,
	 * or a leaveall timer to force clients to re-advertise continued
	 * interest in an attribute.
	 *
	 * BYE   Client detaches from daemon
	 *
	 * M+? - JOIN_MT a MAC address or service declaration
	 * M++   JOIN_IN a MAC Address (XXX: MMRP doesn't use 'New' though?)
	 * M-- - LV a MAC address or service declaration
	 * V+? - JOIN_MT a VID (VLAN ID)
	 * V++ - JOIN_IN a VID (VLAN ID)
	 * V-- - LV a VID (VLAN ID)
	 * S+? - JOIN_MT a Stream
	 * S++ - JOIN_IN a Stream
	 * S-- - LV a Stream
	 *
	 * Outbound messages
	 * ERC - error, unrecognized command
	 * ERP - error, unrecognized parameter
	 * ERI - error, internal
	 * OK+ - success
	 *
	 * Registrar Commands
	 *
	 * M?? - query MMRP Registrar MAC Address database
	 * V?? - query MVRP Registrar VID database
	 * S?? - query MSRP Registrar database
	 *
	 * Registrar Responses (to ?? commands)
	 *
	 * MIN - Registered
	 * MMT - Registered, Empty
	 * MLV - Registered, Leaving
	 * MNE - New attribute notification
	 * MJO - JOIN attribute notification
	 * MLV - LEAVE attribute notification
	 * VIN - Registered
	 * VMT - Registered, Empty
	 * VLV - Registered, Leaving
	 * SIN - Registered
	 * SMT - Registered, Empty
	 * SLV - Registered, Leaving
	 *
	 */
```

Looking through the msrp.c and mvrp.c, I believe the following are valid registrar responses also.
```c
	 * VNE - New attribute notification
	 * VJO - JOIN attribute notification
	 * VLV - LEAVE attribute notification

	 * SNE - New attribute notification
	 * SJO - JOIN attribute notification
	 * SLV - LEAVE attribute notification
```

If you look at the `process_mrp_msg` function in [lib/avtp_pipeline/openavb_common/mrp_client.c](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/openavb_common/mrp_client.c) you can follow the logic to figure out what each of the C=, P=, V=, N=, etc.  stand for.

I *think* C=ID, V=VID, P=Priority, N=Neighbor Priority, D=Substate, S=ID but you should definitely verify for yourself.

It may also be useful to play around with [examples/mrpd/mrpctl.c](https://github.com/AVnu/Open-AVB/blob/master/daemons/mrpd/mrpctl.c).  Its even more simple than the simple_listener/simple_talker in that it only exercises the SRP daemon.

eric-mann, 2016-05-11:
You could ask the guys who actually wrote the sections in the .1Q spec what they were thinking when they came up with the protocol … my interpretation was ‘new’ means creating a new attribute (obviously), ‘join’ means to monitor for the existence of an attribute without necessarily creating a ‘new’ one, ‘in’ is a database state which means the attribute has been created and maintained by someone else, ‘mt’ is another database state which means eMpTy – or that the attribute was created, but is no longer being maintained (a timeout occurred).

So ‘Join’ + ‘In’ means you started to monitor an attribute which exists.

‘Join’ + ‘MT’ means you are looking for an attribute which no one is actively advertising and about to go bye-bye.

This is obviously a gross simplification, and you can consult the relevant sections in the IEEE 802.1Q-2014 spec (Clause 10, in particular understand the state machine in Table 10-3 and Table 10-4 in 10.7.7).

ekm

pinealservo, 2016-05-11:
I find that it helps to think in terms of declaring and registering attributes.  When a port *declares* an attribute, the peer port on the other end of the link notices and *registers* the fact. MRP is actually designed to work in a shared media environment (optimizations for which cause a *lot* of the protocol’s complication) so a registration really means any one or more of the stations on the link have declared the attribute. Ethernet (in its modern AVB-supporting incarnation, anyway) is a point-to-point link protocol, so in the context of Ethernet a registration always refers to the declaration of the single link peer.

The underlying MRP state messages describe both the declarations and registrations that exist at a port. The Join part of the state message (when it exists) says that the port has declared the attribute. The In/MT part of the message describes whether the port has registered the attribute (i.e. noticed a peer’s declaration).

If a port is sending JoinIn messages, then it has both declared and registered the same attribute; i.e. it declared the attribute and noticed that its peer declared the same one.  The only place you should see this in SRP is the Domain attribute, but it’s also seen often in MMRP and MVRP attributes.

If a port is sending JoinMT messages, then it has declared the attribute but has not registered it; i.e. it believes its peer has not declared the same attribute.  This is the normal state of Talker and Listener attributes in SRP when declared by a port, since matches are made when Talker and Listener attributes with the same StreamID value are declared/registered by the same port rather than when declarations/registrations of the *same* attribute occur at a port.

A port sending In messages has not declared the attribute, but has registered it; i.e. it noticed that its peer declared it.  This is the normal state of Talker and Listener attributes in SRP when they’re being received by a port; i.e. a talker will send In for Listener attributes that its link peer declares after it notices that its peer has declared them.

Ports don’t generally send MT messages continuously unless it’s more efficient to encode things that way, since they indicate that the port isn’t declaring the attribute and also hasn’t registered the declaration of a link peer. They’re sent at least once after noticing a peer has withdrawn the attribute, though.

New and Lv are also not sent continuously; they’re indications of a change in declaration state. A New message indicates that the declaration didn’t exist before (it’s essentially a flag on a Join message indicating that it’s a new declaration), and Lv indicates that the declaration is being withdrawn and should be de-registered.

LeaveAll is a special message that prompts re-declaring of all attributes that a port wishes to continue maintaining.  This allows attributes from peers that disappear to be timed-out when they’re not actively withdrawn.

I can’t promise that the command protocol of the MRP daemon matches up with these exactly, but it’s what the state messages in 802.1Q refer to. Don’t sweat the details of the protocol messages too much; you can understand the behavior at the declare/register/propagate level and know enough to get by unless you’re implementing SRP itself or trying to interpret Wireshark traces.

 Looking at msrp.c, the  commands/notifications that start with S and I control the behavior of SRP.  The comment between lines 3563-3573 gives some more details on what the individual S and I commands do; the ones that don’t specifically say which attribute they control are the Talker Advertise controls.

—Levi

andrew-elder, 2016-05-11:
See also
<http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf>

Tigerclaw1993, 2016-05-12:
Tank you for taking the time answering me

That's a lot of information and I will try to go on with this. 
But I am sure there will come up some new questions about AVB and the code after some time so I am going to ask for further help then.

DMats, 2016-05-12:
If your current question is answered, it's probably worth closing this one and asking future questions on new issue threads to keep things organized.

closed 2016-05-17 by Tigerclaw1993