Avnu/OpenAvnu #395: Echo Talker: Stream Attach Callback failed, can anyone tell why?

DMats, 2016-05-09:
I'm attempting to run the echo_listener and echo_talker examples from the Open-AVB-master branch, on an endpoint-to-endpoint network topology.  Both PCs are intel x86_64 architecture, CentOS7 Linux OS, and installed with an Intel I210 card.  

I start the talker first.  Then upon starting the listener, the `strmAttachCb` function is called on the talker side.  Using GDB I've traced the error down to somewhere inside `igb_set_class_bandwidth2`.  I wasn't able to go deeper because I was missing debug symbols for `igb.c` To the best of my knowledge, I believe this function is attempting to initialize the FQTSS credit based traffic shaper hardware on the I210 card.  

Based on the following talker and listener outputs, can anyone tell what I'm doing wrong? Are the input parameters to `igb_set_class_bandwidth2` incorrect in some way?

**Full Talker GDB Output:**
```
[labuser@localhost bin]$ sudo gdb ./openavb_host 
[sudo] password for labuser: 
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-80.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/labuser/Desktop/AVB/Open-AVB-master/lib/avtp_pipeline/build/bin/openavb_host...done.
(gdb) set args -I enp1s0 echo_talker.ini,stream_addr=a0:36:9f:95:5e:c9
(gdb) b setupHWQueue 
Breakpoint 1 at 0x435e76: file /home/labuser/Desktop/AVB/Open-AVB-master/lib/avtp_pipeline/qmgr/openavb_qmgr.c, line 94.
(gdb) run
Starting program: /home/labuser/Desktop/AVB/Open-AVB-master/lib/avtp_pipeline/build/bin/./openavb_host -I enp1s0 echo_talker.ini,stream_addr=a0:36:9f:95:5e:c9
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff7ff8700 (LWP 4202)]
attaching to 0000:02:00.0
[New Thread 0x7ffff4b03700 (LWP 4203)]
[New Thread 0x7fffeffff700 (LWP 4204)]
[New Thread 0x7ffff7ec3700 (LWP 4205)]
[1462833259:894576455 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1462833259:894583382 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1462833259:894599558 OPENAVB osalTime] INFO: local_time = 1462833259805160465
[1462833259:894601203 OPENAVB osalTime] INFO: ml_phoffset = 662162891, ls_phoffset = 2448280
[1462833259:894609106 OPENAVB osalTime] INFO: ml_freqffset = 0.999997, ls_freqoffset = 0.999971
[1462833259:894613941 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.0 (Development)
[1462833259:895054305 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.0 (Development)
[1462833259:896307054 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1462833259:896320792 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1462833260:912015818 OPENAVB Talker] INFO: Register a0:36:9f:95:5e:c9/1: class: B frame size: 118  frame interval: 1
[Switching to Thread 0x7fffeffff700 (LWP 4204)]

Breakpoint 1, setupHWQueue (nClass=1, classBytesPerSec=640000)
    at /home/labuser/Desktop/AVB/Open-AVB-master/lib/avtp_pipeline/qmgr/openavb_qmgr.c:94
94		int err = 0;
(gdb) n
99		if (nClass == SR_CLASS_A) {
(gdb) n
103			class_a_bytes_per_sec =  qmgr_classes[SR_CLASS_A].classBytesPerSec;
(gdb) n
104			class_b_bytes_per_sec = classBytesPerSec;
(gdb) n
107		err = igb_set_class_bandwidth2(qdisc_data.igb_dev, class_a_bytes_per_sec, class_b_bytes_per_sec);
(gdb) p qdisc_data.igb_dev 
$1 = (device_t *) 0x65fc60
(gdb) p *qdisc_data.igb_dev 
$2 = {private_data = 0x65fe00, pci_vendor_id = 32902, pci_device_id = 5427, 
  domain = 0, bus = 2 '\002', dev = 0 '\000', func = 0 '\000'}
(gdb) p class_a_bytes_per_sec 
$3 = 0
(gdb) p class_b_bytes_per_sec 
$4 = 640000
(gdb) n
[1462833267:748280837 OPENAVB Endpoint] INFO: SRP talker callback uid=1: lsnrDecl=2
108		if (err)
(gdb) p err
$5 = 22
(gdb) n
109			AVB_LOGF_ERROR("Adding stream; igb_set_class_bandwidth failed: %s", strerror(err));
(gdb) n
112		return !err;
(gdb) n
113	}
(gdb) n
```

**Full Listener GDB Output:**
```
[labuser@localhost bin]$ sudo gdb ./openavb_host
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-80.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/labuser/Desktop/AVB/Open-AVB-master/lib/avtp_pipeline/build/bin/openavb_host...done.
(gdb) set args -I enp1s0 echo_listener.ini,stream_addr=a0:36:9f:95:5e:c9
(gdb) run
Starting program: /home/labuser/Desktop/AVB/Open-AVB-master/lib/avtp_pipeline/build/bin/./openavb_host -I enp1s0 echo_listener.ini,stream_addr=a0:36:9f:95:5e:c9
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff7ff8700 (LWP 18786)]
attaching to 0000:02:00.0
[New Thread 0x7ffff4b03700 (LWP 18787)]
[New Thread 0x7fffeffff700 (LWP 18788)]
[New Thread 0x7ffff7ec3700 (LWP 18789)]
[1462826915:556649087 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1462826915:556655745 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1462826915:556673383 OPENAVB osalTime] INFO: local_time = 1462826915203952668
[1462826915:556674970 OPENAVB osalTime] INFO: ml_phoffset = 0, ls_phoffset = 240064170
[1462826915:556682601 OPENAVB osalTime] INFO: ml_freqffset = 1.000000, ls_freqoffset = 0.999970
[1462826915:556686608 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.0 (Development)
[1462826915:557134187 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.0 (Development)
[1462826915:558282258 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1462826915:558300438 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1462826915:578369054 OPENAVB Endpoint] INFO: detected domain Class A PRIO=3 VID=0002...
[1462826915:578374389 OPENAVB Endpoint] INFO: detected domain Class B PRIO=2 VID=0002...
[1462826916:567363479 OPENAVB Listener] INFO: Attach a0:36:9f:95:5e:c9/1
[1462826923:905825067 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=1
[1462826924:568097257 OPENAVB Listener] INFO: Starting stream: a0:36:9f:95:5e:c9/1
[1462826924:568107617 OPENAVB Raw Socket] INFO: Using *igb* implementation
[1462826925:576900727] RX UID:1, calls=941, frames=0, lost=-1, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1462826926:576483635] RX UID:1, calls=910, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1462826927:577522125] RX UID:1, calls=918, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1462826928:576638729] RX UID:1, calls=924, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1462826929:576403302] RX UID:1, calls=933, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1462826930:576958522] RX UID:1, calls=902, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1462826931:576662051] RX UID:1, calls=930, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
```

DMats, 2016-05-12:
I've narrowed it down a little more.  I re-built with debug symbols for `igb.c` and I now know that `igb_set_class_bandwidth2` is failing at Line 1445.  After the `ioctl` call, the returned `err` value is 0, and thus, the later lines return `EINVAL = 22`.
```c
	err = ioctl(adapter->ldev, IGB_LINKSPEED, &link);

	if (err) return ENXIO;

	if (0 == link.up) return EINVAL;
```

Before the `ioctl` call:

- `adapter->ldev = 7`
- `IGB_LINKSPEED` is a macro for `_IOW('E', 206, int)`
- `link = {up = 1462069105, speed = 0, duplex = 798067709}`

After the `ioctl` call:
- `link = {up = 0, speed = 0, duplex = 1}`

Does anyone know what this error means?

intel-ethernet, 2016-05-12:
It means the driver doesn’t think link is up (?).

Maybe I missed context of the question

DMats, 2016-05-12:
My apologies.  I probably jumped down into the weeds before fully explaining what I'm doing.  

I'm trying to run the echo_talker and echo_listener using an endpoint to endpoint network topology but I receive some error messages on the talker side and I want to know why they're happening.  I've made sure that my gptp and srp daemons are running on both sides.  

I start the talker first.  It waits for the listener after printing the following line:
```
[1462408169:927527232 OPENAVB Talker] INFO: Register a0:36:9f:95:5e:c9/1: class: B frame size: 118  frame interval: 1
```

Upon starting up the listener, the talker prints the following info and error messages:
```
[1462408179:406759152 OPENAVB Endpoint] INFO: SRP talker callback uid=1: lsnrDecl=2
[1462408179:407015644 OPENAVB QMGR] ERROR: Adding stream; igb_set_class_bandwidth failed: Invalid argument
[1462408179:407019439 OPENAVB Endpoint] ERROR: Error in attach callback: unable to setup stream queues
```

I traced these error messages and they seem to be ultimately caused by an error in the `ioctl` call shown in my previous comment.  

I don't understand why my igb_avb driver would think the link isn't up.  I can successfully ping from one computer to the other.

Is there any other info I could provide that would be helpful?

intel-ethernet, 2016-05-12:
Do you have two I210 cards installed? Is the AVB stack using the I210 you think it should?

DMats, 2016-05-12:
Yes I'm using Intel I210 cards.  Technically I have 2 Intel I210 cards in each PC for a total of 4 I210 cards.

But I'm only using 1 Intel I210 card on each PC right now.  They're connected to each other by a single ethernet cable.  There is no switch in-between.

 I assigned static IP addresses to them and can successfully ping.  In the network settings I see that `enp1s0` ethernet port appears "connected" on both sides.  All other ports are "cable unplugged".

intel-ethernet, 2016-05-12:
I think the Ethernet interface that the avb stack is trying to bind to is the one with the cable unplugged.

Unless the avb stack only connects to the 1st I210 it finds (?).

Can you specify the interface to the AVB stack? (I have never used the echo talker stack personally).

ekm

DMats, 2016-05-12:
Yes I can specify the interface.  I invoke the "echo_talker" using the following commands.  They can also be found in [run_echo_talker.sh](https://github.com/AVnu/Open-AVB/blob/master/run_echo_talker.sh)
```
cd lib/avtp_pipeline/build/bin
exec ./openavb_host -I enp1s0 echo_talker.ini,stream_addr=a0:36:9f:95:5e:c9
```

The -I option allows me to specify the interface, in this case, `enp1s0`

the [echo_talker.ini](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/intf_echo/echo_talker.ini) file contains a variety of config info
Per comments in the .ini file, I override the stream_addr to the MAC address of my talker interface

DMats, 2016-05-12:
**You are right!  I was trying to connect to the wrong I210 card.**  

Using GDB I was able to force `pci_connect` to connect to bus 1 instead of bus 2. For a brief moment, I saw messages on the listener side that I'd never seen before and it was scrolling past very quickly.  Unfortunately then my listener PC crashed and rebooted.  

Crashing aside, I'll count this as a resounding success!  

The logic in [avb.c](https://github.com/AVnu/Open-AVB/blob/master/lib/avtp_pipeline/openavb_common/avb.c) `pci_connect` (shown below) can connect to the wrong device if there are multiple I210 cards.  It does not consider the interface input parameter in it's selection.
```c
	for (dev = pacc->devices; dev; dev = dev->next) {
		pci_fill_info(dev,
				PCI_FILL_IDENT | PCI_FILL_BASES | PCI_FILL_CLASS);
		igb_dev->pci_vendor_id = dev->vendor_id;
		igb_dev->pci_device_id = dev->device_id;
		igb_dev->domain = dev->domain;
		igb_dev->bus = dev->bus;
		igb_dev->dev = dev->dev;
		igb_dev->func = dev->func;
		snprintf(devpath, IGB_BIND_NAMESZ, "%04x:%02x:%02x.%d",
				dev->domain, dev->bus, dev->dev, dev->func);
		err = igb_probe(igb_dev);
		if (err)
			continue;

		printf("attaching to %s\n", devpath);
		err = igb_attach(devpath, igb_dev);
		if (err) {
			printf("attach failed! (%s)\n", strerror(err));
			continue;
		}

		err = igb_attach_tx(igb_dev);
		if (err) {
			printf("attach_tx failed! (%s)\n", strerror(err));
			continue;
		}

		goto out;
	}
```
The same error exists in [examples/common/avb.c](https://github.com/AVnu/Open-AVB/blob/master/examples/common/avb.c)

DMats, 2016-08-06:
See #438 for additional info.

Leon7801, 2024-10-27:
Hello everyone)) I must say right away - I'm not really a programmer. Please tell me how to make Intel I250 work on AVB in Windows 11? I have a presonus studio live 32r digital remote control and I need to connect a PC via avb, thank you!!!