Avnu/OpenAvnu #398: DeviceIoControl error code 50

ibrahimdanish, 2016-05-16:
Hi,

I am doing some experimental work with OpenAVB gptp daemon for windows. I have compiled the solution in VS2015. When I run the code I see the following messages with error 50 in the console. I tried to debug the code and found that DeviceIoControl is returning error in the class WindowsTimestamper in windows_hal.hpp. Does anyone know the reason? I have intel I218-LM card in my machine.

Opening: rpcap://\Device\NPF_{DEFFEC15-3336-420D-A649-25B03465774E}
Adapter UID: {DEFFEC15-3336-420D-A649-25B03465774E}
mhz: 2294
Multiplier: 23
Frequency: 2300000000
Starting PDelay
Error is: 50
Error is: 50
Error is: 50
Error is: 50
Error is: 50
Error is: 50
Error (TX) timestamping PDelay request (Retrying-0), error=-1
Error is: 50
Invalid TX
ERROR at 680 in c:\projectarea\open-avb-master\open-avb-master\daemons\gptp\comm
on\ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-1

Regards,

Ibrahim

christopher-s-hall, 2016-05-16:
You’ll need to make some registry changes.

First find the driver key. To do this go to device manager, device properties, details, and select driver key.

On my machine, for example, it’s: {4d36e972-e325-11ce-bfc1-08002be10318}\0000. I would, search the registry for 4d36e972-e325-11ce-bfc1-08002be10318 where there is a subkey 0000. On my machine it’s in HKLM/System/ControlSet001/Control/Class.

Now add a DWORD value called TimeSync with a value of 1 to the subkey (0000 in my example). Reset the driver by disabling and re-enabling (or reboot) and it should work. Make sure you’re using up to date drivers from the Intel website (this is a private driver interface). Version >= 20.3 is fine.

Chris

andrew-elder, 2016-05-16:
Thanks Chris. Is http://www.intel.com/content/www/us/en/support/network-and-i-o/ethernet-products/000005504.html the correct place to grab drivers from?

christopher-s-hall, 2016-05-16:
Yes. These are the right drivers.

ibrahimdanish, 2016-05-17:
Thanks a lot Chris, it worked :)

Regards,

Ibrahim

davidcemin, 2016-05-17:
looks like it is working. Can we close this?

andrew-elder, 2016-05-17:
No, we need to add this information to the top level readme so the issue does not come up again. @davidcemin - do you want to do that?

davidcemin, 2016-05-17:
sure! Im assigning it to myself.

ibrahimdanish, 2016-05-18:
Hi,

You can close it @davidcemin if you want.

Regards,

Ibrahim

davidcemin, 2016-05-18:
@ibrahimdanish I will keep it open until we put the information about the registry changes provided by ChrisH on the toplevel Readme file.
Thank you.

closed 2016-05-21 by davidcemin