Avnu/OpenAvnu #401: AsCapable unstable

RikusW, 2016-05-17:
Everything worked fine on the previous version but it is unstable on the new one.
I'm using a XMOS XR-AVB-LR-BRD, it does the same when using a firefox board.
git diff 5f44c4e..32dab06

sudo Open-AVB/daemons/gptp/linux/build/obj/daemon_cl eth1 &
sudo Open-AVB/daemons/mrpd/mrpd -mvs -i eth1 &
sudo Open-AVB/daemons/maap/linux/maap_daemon -i eth1 &

====The output of the previous version====

ERROR at 636 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay

SENT MAAP_PROBE 
process_events()

SENT MAAP_PROBE 
AsCapable: Enabled

SENT MAAP_PROBE 

SENT MAAP_ANNOUNCE 

ANNOUNCED ADDRESS:
0x91 0xe0 0xf0 0 0x53 0xc2 

STATE change to DEFEND:
*** Announce Timeout Expired - Becoming Master
New Grandmaster "A0:36:9F:FF:FE:53:C1:E9" (previous "00:00:00:00:00:00:00:00")

====The output of the new version====

ERROR at 730 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay

SENT MAAP_PROBE 
process_events()

SENT MAAP_PROBE 
AsCapable: Enabled
AsCapable: Disabled

SENT MAAP_PROBE 

SENT MAAP_ANNOUNCE 

ANNOUNCED ADDRESS:
0x91 0xe0 0xf0 0 0x79 0xd4 

STATE change to DEFEND:
AsCapable: Enabled
AsCapable: Disabled
*** Announce Timeout Expired - Becoming Master
New Grandmaster "A0:36:9F:FF:FE:53:C1:E9" (previous "00:00:00:00:00:00:00:00")
AsCapable: Enabled
AsCapable: Disabled
AsCapable: Enabled
AsCapable: Disabled
AsCapable: Enabled
AsCapable: Disabled
.AsCapable: Enabled
AsCapable: Disabled
This continues endlessly.

andrew-elder, 2016-05-17:
What branch are you using?

RikusW, 2016-05-17:
master -- old working 5f44c4e  new 32dab06

pinealservo, 2016-05-17:
This is probably the neighbor prop thresh delay issue.  I have a fix for it that will be in my next pull request.

Are you hooking them back-to-back or do you have a non-AVB peeker/bridge in-between?

RikusW, 2016-05-18:
The i210 is directly connected to the XMOS / Firefox

andrew-elder, 2016-05-18:
I can confirm similar results with Win10 and i217-LM (built from open-avb-next). Log follows....

```
ERROR at 136 in D:\asi\sw\Open-AVB\daemons\gptp\windows\daemon_cl\windows_hal.cpp: Warning: Mbit receive PHY delay is unknown using 0
ERROR at 140 in D:\asi\sw\Open-AVB\daemons\gptp\windows\daemon_cl\windows_hal.cpp: Warning: Mbit transmit PHY delay is unknown using 0
Starting PDelay
ERROR at 141 in D:\asi\sw\Open-AVB\daemons\gptp\common\ptp_message.cpp: *** Received an event packet but cannot retrieve timestamp, discarding. messageType=3,error=-72

ERROR at 352 in D:\asi\sw\Open-AVB\daemons\gptp\common\ieee1588port.cpp: Discarding invalid message
ERROR at 1412 in D:\asi\sw\Open-AVB\daemons\gptp\common\ptp_message.cpp: >>> Received PDelay followup but no RESPONSE exists
AsCapable: Enabled
*** Announce Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:00:00:00:00:00:00:00")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
AsCapable: Enabled
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
AsCapable: Enabled
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
Switching to Master
Switching to Slave
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
AsCapable: Enabled
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
AsCapable: Enabled
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FFOpening: rpcap://\Device\NPF_{91D2E740-71E1-49A7-834A-17D868532C98}
:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
AsCapable: Disabled
AsCapable: Enabled
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
*** Sync Timeout Expired - Becoming Master
New Grandmaster "F8:B1:56:FF:FE:E4:0E:DA" (previous "00:04:96:FF:FE:8B:C9:3F")
Switching to Slave
New Grandmaster "00:04:96:FF:FE:8B:C9:3F" (previous "F8:B1:56:FF:FE:E4:0E:DA")
```

intel-ethernet, 2016-05-18:
It has been a while since I’ve looked at the I217 design + Windows driver implementation … but I suspect a power-saving mode is enabled which is probably adding error to the packet timestamps. Specifically, there is a power-save mode where the link between the MAC and the PHY is put into a low power state. As the timestamps are taken in the MAC, and it could take 10’s of usec to bring the link out of an idle state, the resulting timestamps would be useless from an 802.1AS perspective.

To test this hypothesis, you can run background network traffic to keep the MAC/PHY from entering this power-save mode, and see if the gPTP daemon maintains asCapable.

Fwiw.

ekm

christopher-s-hall, 2016-05-18:
It's possible that this is a problem, but I've used the hardware for 802.1AS and it (barely in some cases) worked. The timestamp jitter was in the range of hundreds of nanoseconds which might cause problems with the pdelay threshold, but I was always able to tune it. The best way to get a handle on the problem is just to print the link delay - maybe we need a command line switch for this(?)

andrew-elder, 2016-05-18:
I tested with ping traffic through the i217 at the same time gPTP was running. No change the results. @christopher-s-hall, what setup do you use for testing the Windows build?
I am connected to an Extreme switch. I could try an XMOS device I guess.
I'll see about getting link delay output, but most likely it won't be until Friday....

pinealservo, 2016-05-18:
If you run my automotive profile branch, it'll print out the link delay and
the threshold when it drops asCapable due to the link delay exceeding the
threshold.

RikusW, 2016-05-23:
The automotive patch fixed this.

closed 2016-05-23 by RikusW