Avnu/OpenAvnu #403: IRC channel #openavb created on irc.freenode.net

RikusW, 2016-05-17:
I'll give operator access to AVnu/Open-AVB developers, just ask.

RikusW, 2016-05-17:
It is preferable that you register you nick in NickServ
/msg NickServ help
/msg NickServ help register

RikusW, 2016-05-18:
Please use your GitHub username on IRC to avoid confusion, unless you already have a freenode account.

andrew-elder, 2016-05-18:
@RikusW - can you elaborate on what using irc.freenode.net will do for the Open-AVB project. I'm missing the "why do I care about this" part. It may be obvious to others, but it isn't to me. Thanks.

RikusW, 2016-05-18:
IRC allow for realtime discussion so it is faster than the forums.
It is more informal and allows for general discussion and planning of the project.
The questions asked on the channel will help those doing documentation to improve it.

pinealservo, 2016-05-24:
I will be idling there most of the time, since I already use IRC on freenode for other purposes. Since this isn't actually an issue as posted, I'm going to close it.

If there's a desire to promote it more widely, a feature request issue for a project page change or a pull request implementing such would be a better place to discuss it.

closed 2016-05-24 by pinealservo