Avnu/OpenAvnu #404: gPTP: missing default PHY delay values for Windows build

andrew-elder, 2016-05-17:
This issue applies to open-avb-next.

The Windows HAL layer is missing default PHY delay values for the I210 and is missing default 1G delay values for all adapter types.

davidcemin, 2016-05-17:
Is this something that I forgot to implement in one of my patches? If yes, please assign this issue to me.

christopher-s-hall, 2016-05-17:
I changed this. The default value is zero. Previously they were I210 values. I don't think these numbers make sense because there aren't any drivers that implement this interface and support the I210 hardware.

pinealservo, 2016-05-17:
Presumably @andrew-elder has some sort of hardware/driver combination that's usable for AVB, yet doesn't get values initialized by the code you added to specify delays.

What do you mean by "this interface"?  Do you mean the hardware timestamp interface used by gPTP, or do you mean something specific to querying the driver to figure out how to set the delay numbers?

I think an auto-detection mechanism for it is great, but if it's not universally supported, how many options are there that support the timestamping but not the auto-detection?  Maybe we can build on the table of hardware and add a single command line flag to specify the NIC (and thus the set of values corresponding to it in the table) to use instead of requiring the user to know and manually specify the values.

And what about the values for the hardware entries that are currently in the table, but don't currently have all the delay values filled out?  Are they just not available?  Do we have to measure them somehow?

christopher-s-hall, 2016-05-17:
Below…

What do you mean by "this interface"? >>Intel’s private driver API for timestamping

Do you mean the hardware timestamp interface used by gPTP, or do you mean something specific to querying the driver to figure out how to set the delay numbers? >>There isn’t any query to the driver to figure this out. It’s a table lookup.

Do we have to measure them somehow? >> Without a GPIO output I don’t think there’s any way to do that.

andrew-elder, 2016-05-17:
I'm testing on a Dell Windows 10 machine that has a I217-LM on the motherboard. ie all code paths I'm reporting on are those publicly available in Open-AVB (except of course Intels IGB driver for Windows).

The PHY delay values for I217 at 1G are not currently filled in. Can we at least do that? Aren't the numbers part of the chip datasheet?

The auto detect method for the NIC interface uses Windows OS calls to loop over available interfaces and compare the string name for the interface against the list of "known" interfaces. So I think that for windows this lookup method by adapter type could be considered to be "universally supported".

pinealservo, 2016-05-17:
So I'd be out of luck trying to do Windows testing with my i210 cards?  I guess I need to figure out if I've got a machine with one of the supported interfaces.

IIRC, not all of the numbers we currently have for the i210 are in the current datasheet.  There's some discussion in the issues somewhere regarding where they came from.

christopher-s-hall, 2016-05-17:
Below…

The code is here:

static DevicePhyDelayMapping DevicePhyDelayMap[] =

{

       {{ -1, -1, 6950, 8050, }, I217_DESC },

       {{ -1, -1, 6700, 7750, }, I219_DESC },

       {{ -1, -1, -1, -1 }, NULL },

};

struct phy_delay

{

   int mb_tx_phy_delay;

   int mb_rx_phy_delay;

   int gb_tx_phy_delay;

   int gb_rx_phy_delay;

};

Can you tell me what the name of the interface in device manager is? Maybe there’s an extra letter added or remove… which may or may not be consequential to the PHY delay/

Aren't the numbers part of the chip datasheet? >> No.

andrew-elder, 2016-05-18:
Ah, so the GB fields are the ones filled in! I'll check in the morning.

andrew-elder, 2016-05-18:
I can confirm that the Gbit fields are filled in correctly for the I217 and that the I217 is "discovered" correctly by the network interface scanning code in the Windows HAL module. I was confused because the lack of 100Mbit values caused error output, but I guess those values (of 0) are not being used since I am running at 1G.

I'm ok closing this issue for the moment.

christopher-s-hall, 2016-05-18:
We still need link speed detection. For now, the code assumes 1G link.

closed 2016-05-24 by pinealservo