Avnu/OpenAvnu #407: [Question] Support for TSN

bajantri, 2016-05-18:
Hi Everyone,

Recently I heard about the TSN working group (Formerly known as AVB)
introducing new set of clauses/standards for AVB such as,

802.1AS-Revision, 802.1Qbu, 802.1Qbv, 802.1Qcc, 802.1CB, 802.1Qch, 802.1Qci.

All above mentioned TSN standards are still under development process
and are not available publicly.

With this small background, I have few queries.

1. Is there a support for any of the above mentioned standards in the Open-AVB?

2. Is the Open-AVB source code scalable to add support for above
mentioned standards (software part) ?

If this is not a correct forum to post such question please point me
to the appropriate forum.

Thanks,
Praveen Bajantri
Vayavya Labs.

pinealservo, 2016-05-18:
The new revisions to the standards related to AVB/TSN should largely remain backward-compatible with the current ones, so in that sense OpenAVB does support them. Once they are complete, they just become part of 802.1AS and 802.1Q. But there's not been an explicit attempt to track and implement the in-progress changes to the standards aside from perhaps a few things that have been needed immediately by users, so OpenAVB does not currently support any new features or requirements unless they happen to have been needed by some developers.

It should definitely be possible to extend the OpenAVB code to support changes in the standards, but I don't know if anyone currently has plans to do so. It will become more likely as the standards changes are finalized and incorporated into industry certification programs such as those supplied by the AVnu Alliance, which provides support for maintaining the code.

bajantri, 2016-05-19:
Thanks for the quick response.

closed 2016-05-24 by pinealservo