Avnu/OpenAvnu #409: pci_connect seems to be duplicated 5 times...

RikusW, 2016-05-19:
While fixing #396 I discovered the duplication in the following files:

daemons/gptp/linux/src/linux_hal_i210.cpp
examples/jackd-talker/jackd_talker.c
examples/simple_talker/simple_talker.c
examples/common/avb.c
lib/avtp_pipeline/openavb_common/avb.c

RikusW, 2016-05-20:
I reformatted all for easy comparison and submitted #408

RikusW, 2016-05-27:
Should pci_connect be removed from examples/jackd-talker/jackd_talker.c and examples/simple_talker/simple_talker.c and instead use the one in examples/common/avb.c ?

RikusW, 2016-05-27:
Reduce to 3 duplications with #416

pinealservo, 2016-06-28:
@RikusW Do you consider this fixed now, or is there still more to do?

RikusW, 2016-06-29:
Then avb.c/h is done, the other examples/common files can be done in a separate branch.

pinealservo, 2016-07-26:
I finally looked at this other instance; it's using the interface for PPS configuration rather than packet handling, so it doesn't nead igb_attach_tx. It's probably not worth the effort to factor out the common code right now, so I'm going to close the issue.

closed 2016-07-26 by pinealservo