Avnu/OpenAvnu #412: gPTP: should log level be set at compile time instead of runtime?

andrew-elder, 2016-05-20:
Currently the gPTP log level is set at compile time. I've opened this issue to discuss whether it would be better to have the log level be a run-time option.  The original implementer obviously thought compile time was a good idea for some reason? Comments....

intel-ethernet, 2016-05-20:
Compile-time enables creating a smaller binary (no unnecessary logging/debug strings).

davidcemin, 2016-05-20:
Makes sense, I agree that there is no need to keep these symbols if not in use. Is the difference in size really relevant if we are not running this on a device with limited storage (typical in embedded world) ? (Sorry, but I dont know if we run this in platforms different than PCs).

intel-ethernet, 2016-05-20:
The OpenAVB gPTP daemon is running in the dash of some luxury automobiles right now …

andrew-elder, 2016-05-20:
The most flexible arrangement would be to have BOTH compile time #defines and a run-time argument. I think a run-time configurable argument/variable is sometimes useful. Thanks for your input Eric.

davidcemin, 2016-05-20:
Thats nice to know about open-avb running in some cars, really cool !  I see benefits in having both as well., but we would have to be aware of the size as previously said a couple of comments above. Hopefully it won't increase too much the size of gPTP.

pinealservo, 2016-05-24:
The implementation function currently has a "level" parameter that is a string, which directly provides the "level" portion of the string output.  That could be changed to some sort of level id that could be used to both look up the correct string describing the level or serve as a run-time filter.  

Enabling all the level-specific logging macros and setting the run-time filter value below to something other than "show all", however that would be encoded, would allow run-time configuration; Leaving it at "show all" and disabling unwanted levels via the level-specific macro facility would provide compile-time configuration.

If we want to do this, it will require:

1. Updating the log function, perhaps as I described above.
2. Updating the macros to use the new log function interface.
3. Providing a global "log level" variable and a default value for it, along with some way to alter the value it takes at startup and possibly at arbitrary points during execution via a control socket or something.

If that sounds reasonable, we should probably update this from a question to a feature request, or create a new feature request and close this.

andrew-elder, 2016-05-24:
Updated issue with "enhancement" label. I think we should look at optional runitime log level setting.