Avnu/OpenAvnu #414: Out-of-bounds read in ClockIdentity::set()

m46kruk, 2016-05-25:
Hello everyone,

due to improper message-length validation, it is possible for an attacker to cause the gptp daemon to die with a segmentation fault. This happens because the length of an announce message's optional Path Trace TLV is not properly validated in `buildPTPMessage()`.
On top of that, `PathTraceTLV::parseClockIdentity()` relies on the unchecked value of the length field and therefore blindly passes a pointer (that may or may not point to a valid identity) to `ClockIdentity::set()`.

I have attached the code for an application that sends a malformed message with an invalid length (0xFFFF) in the Path Trace TLV's length, which reliably caused daemon_cl to die with a segmentation fault on my side.

I am not sure what the proper process for this is, but I have committed a patch to my fork of this repository. Should I also create a pull request?

[pt_vuln_exploit.zip](https://github.com/AVnu/Open-AVB/files/281947/pt_vuln_exploit.zip)

andrew-elder, 2016-05-25:
Assuming you have based your fix on the current code in AVnu branch open-avb-next, please do create a pull request to AVnu open-avb-next so that folks can review it. Thanks for finding this and double thanks for fixing it!

m46kruk, 2016-05-25:
My patch was based on master, but I've fixed it - it's now based on open-avb-new. 
I'm creating the pull request now.

closed 2016-05-26 by pinealservo