Avnu/OpenAvnu #418: Link not asCapable on i210

RikusW, 2016-05-30:
I start the daemons using run_daemons and use it with a Marvell Firefox board with kernel and Marvell patch version FFxAV 2.6.30.10 #1.0.00266
Same thing when using a XMOS board.

STATUS   : GPTP [16:39:49:442] Starting PDelay

SENT MAAP_PROBE 
process_events()

SENT MAAP_PROBE 

SENT MAAP_PROBE 

SENT MAAP_ANNOUNCE 

ANNOUNCED ADDRESS:
0x91 0xe0 0xf0 0 0xef 0xe7 

STATE change to DEFEND:
STATUS   : GPTP [16:39:52:442] *** Announce Timeout Expired - Becoming Master

STATUS   : GPTP [16:39:52:443] New Grandmaster "A0:36:9F:FF:FE:53:C1:E9" (previous "00:00:00:00:00:00:00:00")

SENT MAAP_ANNOUNCE

RikusW, 2016-06-06:
I changed neighborPropDelayThresh to 3000 and now it is asCapable at a delay of 1360ns.
The asCapable status is still not printed to the terminal.

pinealservo, 2016-07-26:
The asCapable status changes are printing now; please open another issue if there are remaining problems.

closed 2016-07-26 by pinealservo