Avnu/OpenAvnu #419: igb_set_class_bandwidth vs. igb_set_class_bandwidth2

DMats, 2016-06-06:
What is the difference between igb_set_class_bandwidth and igb_set_class_bandwidth2?  Under what circumstances would you use each?

intel-ethernet, 2016-06-08:
igb_set_class_bandwidth takes the same parameters you would configure SRP to allocate bandwidth (TPKT_SZ, Class, Packets per Class Observation Interval, etc).

IIRC, someone submitted igb_set_class_bandwidth2 as part of the avtp submission. I believe the intent was to specify a bytes-per-second interface rather than max packet size x packet rate.

I guess it doesn’t matter so long as the end result is the same.

ekm

DMats, 2016-06-08:
Thank you for your explanation.  This helps.

closed 2016-06-08 by DMats

MarcinMiklas, 2016-09-12:
I wrote igb_set_class_bandwidth2. It was added to support multiple talker streams. If two takers uses different packet size it was impossible to precisely allocate required bandwith using igb_set_class_bandwidth. That's why igb_set_class_bandwidth2 was introduced, which allows to specify bytes-per-second for each traffic class.