Avnu/OpenAvnu #421: Question regarding the Ethernet Adapter necassary to run gPTP on a Linux PC.

pavan-pan, 2016-06-07:
Hi,

I am new to Open AVB. Currently, I am planning to implement gPTP on my Linux PC. My PC has a Qualcomm Atheros 8171/8175 Ethernet adapter. 

Since the igb drivers are only supported on Intel cards, I am currently not sure if i can implement gPTP on my Linux machine. It would be of great help if someone could just clarify and also point me to the basic hardware requirements necessary to implement the gPTP on a Linux PC. I am also planning to port it later to a raspberry pi or a ARM board.

I tried to compile the gPTP and run it on my Linux PC and I am getting the below error.

ERROR at 361 in ../src/linux_hal_generic.cpp: Failed to configure timestamping: Operation not supported
ERROR at 902 in ../src/linux_hal_common.cpp: post_init failed

Does that mean that my assumption is right ?

Regards,
Pavan.

m46kruk, 2016-06-07:
Hey Pavan,

your assumption is correct, in that gPTP will not work with your Ethernet adapter. In order to achieve (below) microsecond precision in PTP implementations, hardware support is necessary for the timestamping of messages. To my knowledge, this is only supported by Intel i210-family Ethernet adapters at the moment.

As far as I'm concerned, writing a patch to perform the timestamping in software is a possibility in order to port gPTP to devices such as the Raspberry Pi, but the resulting degradation of precision would probably defeat the purpose of using PTP in the first place.

Best regards
Matt

intel-ethernet, 2016-06-08:
Here’s a list of adapters from linuxptp.sourceforge.net (ptp4l) which support HW timestamping – it may not be an exhaustive list. It doesn’t list the I210 for example, nor the integrated LANs found in various client PCH chipsets (e.g. the I219).

You may also need to configure and recompile the kernel to enable PTP, which will activate PTP in the Ethernet drivers.

ekm

Driver

Hardware

Version

amd-xgbe

AMD 10GbE Ethernet Soc

3.17

bfin_mac

Analog Blackfin

3.8

bnx2x

Broadcom NetXtremeII 10G

3.18

cpts

Texas Instruments am335x

3.8

e1000e

Intel 82574, 82583

3.9

fm10k

Intel FM10000

3.18

fec

Freescale i.mx6

3.8

gianfar

Freescale eTSEC PowerPC

3.0

i40e

Intel XL710 Family

3.14

igb

Intel 82576, 82580

3.5

ixgbe

Intel 82599

3.5

mlx4

Mellanox 40G PCI

3.14

ptp_ixp46x

Intel IXP465

3.0

ptp_phc

Lapis EG20T PCH

3.5

sfc

Solarflare SFC9000

3.7

stmmac

STM Synopsys IP Core

3.10

tg3

Broadcom Tigon3 PCI

3.8

tilegx

Tilera GBE/XGBE

3.12

Fwiw.
ekm

pavan-pan, 2016-06-08:
Thank Matt and ekm for your kind answers. My doubt's have been cleared.

closed 2016-06-08 by pavan-pan