Avnu/OpenAvnu #427: AVB Talker understanding

amiguptaa2000, 2016-06-28:
Hi,
I am a beginner for AVB protocols. I am going through 802.1Qav, 802.1Qat protocols.
I have a question on talker behavior. 
In my thinking that a talker can send 1 stream at a time and BW is known to talker by SRP,
than how/why there is queuing happening per-stream and per-traffic class.

Please guide me for the same asap.

Thanks in advance

jpolin, 2016-07-19:
802.1Qat: During SRP, talker(s) request to send a certain amount of data (per specified class interval) to end points that have identified themselves as listeners. If all intermediary links can support the bandwidth requested by the talker, then all participating parties are aware of the stream being successfully established. The talker can then confidently send that much data to listeners. It is possible that more listeners try to join but cannot due to the increased bandwidth that would require (will only matter in the case of multiple streams).

802.1Qav: Switches do their credit shaping by port. Therefore, if a switch has 2 class-A streams coming in through the same port, and expects 2 packets-per-interval on each stream, then it will be stream-agnostic when applying credit shaping. I believe TSN, to which AVB is a precursor, will credit shape by stream rather than port.

pinealservo, 2016-07-21:
The traffic shaping in AVB actually happens per traffic queue on the port, not globally per port.  Each port can have multiple queues, which have been used for various sorts of priority-differentiated service for a long time. 

Current implementations do not differentiate between multiple streams in the same traffic class/queue; the reservations for the class are just summed to give the total credits per interval in the credit-based shaper. I don't know off the top of my head how future revisions of 802.1Q will be different.

amiguptaa2000, 2016-07-21:
Hi @jpolin and @pinealservo ,

Thank you very much for your valuable information. I will try to understand the AVB protocols based on this and will try to get help if still some understanding issue left.

closed 2017-02-13 by andrew-elder