Avnu/OpenAvnu #429: simple_talker problem

stujeffrey016, 2016-07-04:
Dear all,
I'm a student who interesting in AVB.
I want to build up a simple_talker and listener platform and do some reach.
Can you guys please help me solve my error?
intel i210<===>Switch<===>intel i210
I use VMware and install Ubuntu 14.04 LTS as my OS.

[cmake .. -G "Unix Makefiles"
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:4 (add_subdirectory):
  The source directory

    /home/jeffreyo/Open-AVB-open-avb-next/thirdparty/cpputest

  does not contain a CMakeLists.txt file.

CMake Warning (dev) at daemons/mrpd/tests/simple/CMakeLists.txt:19 (link_directories):
  This command specifies the relative path

    mrpd_simple_test

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at daemons/mrpd/tests/simple/CMakeLists.txt:19 (link_directories):
  This command specifies the relative path

    ../../../../thirdparty/cpputest/src/CppUTest

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at daemons/mrpd/tests/simple/CMakeLists.txt:19 (link_directories):
  This command specifies the relative path

    ../../../../thirdparty/cpputest/src/CppUTestExt

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
](url)

I have used these commands
rmmod igb
sudo modprobe i2c_algo_bit
sudo modprobe dca
sudo modprobe ptp
sudo insmod ./igb_avb.ko

[[ 3621.481352] igb_avb: module verification failed: signature and/or required key missing - tainting kernel
[ 3621.481521] igb_avb: Unknown symbol dca_remove_requester (err 0)
[ 3621.481579] igb_avb: Unknown symbol ptp_clock_index (err 0)
[ 3621.481692] igb_avb: Unknown symbol dca_add_requester (err 0)
[ 3621.481704] igb_avb: Unknown symbol ptp_clock_register (err 0)
[ 3621.481738] igb_avb: Unknown symbol dca_unregister_notify (err 0)
[ 3621.481752] igb_avb: Unknown symbol dca_register_notify (err 0)
[ 3621.481767] igb_avb: Unknown symbol dca3_get_tag (err 0)
[ 3621.481779] igb_avb: Unknown symbol ptp_clock_unregister (err 0)
[ 3808.912052] igb_avb: Unknown symbol dca_remove_requester (err 0)
[ 3808.912098] igb_avb: Unknown symbol ptp_clock_index (err 0)
[ 3808.912172] igb_avb: Unknown symbol dca_add_requester (err 0)
[ 3808.912181] igb_avb: Unknown symbol ptp_clock_register (err 0)
[ 3808.912210] igb_avb: Unknown symbol dca_unregister_notify (err 0)
[ 3808.912221] igb_avb: Unknown symbol dca_register_notify (err 0)
[ 3808.912234] igb_avb: Unknown symbol dca3_get_tag (err 0)
[ 3808.912243] igb_avb: Unknown symbol ptp_clock_unregister (err 0)
[ 5007.486425] igb_avb: Unknown symbol dca_remove_requester (err 0)
[ 5007.486528] igb_avb: Unknown symbol ptp_clock_index (err 0)
[ 5007.486688] igb_avb: Unknown symbol dca_add_requester (err 0)
[ 5007.486709] igb_avb: Unknown symbol ptp_clock_register (err 0)
[ 5007.486773] igb_avb: Unknown symbol dca_unregister_notify (err 0)
[ 5007.486799] igb_avb: Unknown symbol dca_register_notify (err 0)
[ 5007.486826] igb_avb: Unknown symbol dca3_get_tag (err 0)
[ 5007.486849] igb_avb: Unknown symbol ptp_clock_unregister (err 0)
[ 5014.672468] igb_avb: Unknown symbol dca_remove_requester (err 0)
[ 5014.672605] igb_avb: Unknown symbol ptp_clock_index (err 0)
[ 5014.672813] igb_avb: Unknown symbol dca_add_requester (err 0)
[ 5014.672840] igb_avb: Unknown symbol ptp_clock_register (err 0)
[ 5014.672921] igb_avb: Unknown symbol dca_unregister_notify (err 0)
[ 5014.672953] igb_avb: Unknown symbol dca_register_notify (err 0)
[ 5014.672988] igb_avb: Unknown symbol dca3_get_tag (err 0)
[ 5014.673016] igb_avb: Unknown symbol ptp_clock_unregister (err 0)
[ 5335.960441] dca service started, version 1.12.1
[ 5339.634759] pps_core: LinuxPPS API ver. 1 registered
[ 5339.634763] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[ 5339.638414] PTP clock support registered
[ 5353.805456] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[ 5353.805459] Copyright (c) 2007-2015 Intel Corporation.
[ 5992.911970] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[ 5992.911977] Copyright (c) 2007-2015 Intel Corporation.
[ 8253.791488] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[ 8253.791492] Copyright (c) 2007-2015 Intel Corporation.
[ 9230.598197] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[ 9230.598201] Copyright (c) 2007-2015 Intel Corporation.
[ 9827.921115] systemd-hostnamed[108073]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[16185.819973] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[16185.819980] Copyright (c) 2007-2015 Intel Corporation.
[17638.864564] systemd-hostnamed[118424]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[17862.631595] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[17862.631602] Copyright (c) 2007-2015 Intel Corporation.
[19195.314473] systemd-hostnamed[123163]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
](url)

andrew-elder, 2016-07-05:
> Rodolfo Giometti <url>giometti@linux.it <mailto:giometti@linux.it>

Did you do
git submodule update --recursive
?

You can also read .travis.yml and travis.sh for build steps.

pinealservo, 2016-07-26:
@stujeffrey016 Any update on this?  Were you able to get it to build?

stujeffrey016, 2016-07-28:
@andrew-elder 
@pinealservo 
I changed to the real OS ubuntu 14.04 LTS, and it works.
I think the problem is that my VMware didn't connect to intel I210 card, and I don't know how to setup my environment, so I gave up and use the real OS to do my experiments. 
Thank you guys.

closed 2016-07-28 by andrew-elder

117104010497, 2021-04-02:
modprobe ptp
https://ask.csdn.net/questions/1903222