Avnu/OpenAvnu #432: Automotive Profile gPTP Sync Intervals and PDelay Request Intervals

DMats, 2016-07-14:
Take a glance at the if statement located at:  [line 108 of ieee1588port.cpp](https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/ieee1588port.cpp#L108).  I'm not 100% sure, but it appears that **the values for initialLogPDelayReqInterval and operLogSyncInterval seem to be switched**.

For reference, see Table 12 and 13 (section 6.2.6) from the [Automotive Ethernet AVB Functional and Interoperability Specification Revision 1.4](http://avnu.org/wp-content/uploads/2014/05/AutoCDSFunctionalSpec-1_4-public_with_legal_notices.pdf).

1. On [line 115](https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/ieee1588port.cpp#L115) and [line 130](
https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/ieee1588port.cpp#L130):
initialLogPDelayReqInterval is assigned 125ms despite the 1s minimum defined in Table 12 and 13.

2. On [line 121](https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/ieee1588port.cpp#L121) and [line 136](https://github.com/AVnu/Open-AVB/blob/feature-gptp-avnu-automotive-profile/daemons/gptp/common/ieee1588port.cpp#L136):
operLogSyncInterval is assigned 1s which is the maximum value defined in Table 12 and 13.  This is not technically incorrect and may have been intentional, but I figured I'd call it out just in case.

kencarlino, 2016-07-14:
@DMats For your item 1 you are right that is a error. For item 2 it was intentional to have a default of 1 second for the Operation Sync rate.   

Note: The automotive profile functionality has been merged into open-avb-next and this item 1 has been corrected there earlier in this week. There are also some other changes/fixed that have also been applied to the automotive profile functionality in the open-avb-next branch.  I personally don't have plans to do further updates to the feature-gptp-avnu-automotive-profile branch that you are referencing. 

Also note: There will be will another upcoming pull request for some more automotive profile improvements/fixes within the next week into the open-avb-next branch.

DMats, 2016-07-14:
Oh! My apologies.  I wasn't watching the commits closely enough.  Thank you for your prompt response.  I'll look at the open-avb-next branch instead.  My question is answered.

closed 2016-07-14 by DMats