Avnu/OpenAvnu #433: Question:  AVB Automotive Profile capable switch

DMats, 2016-07-14:
Are there any commercially available switches capable of the AVB Automotive Profile gPTP behavior explained in section 6.3 of the [Automotive Ethernet AVB Functional and Interoperability Specification Rev 1.4](http://avnu.org/wp-content/uploads/2014/05/AutoCDSFunctionalSpec-1_4-public_with_legal_notices.pdf)

1. AED-B Sync processing during loss of sync:
2. AED-B Sync Recovery:
3. AED-B Sync Absence:
4. AED-B Fixed GM Information:

I've seen the NXP offering: http://www.nxp.com/products/interface-and-connectivity/wired-connectivity/ethernet/5-port-automotive-ethernet-switch:SJA1105TEL but it's unclear to me whether it supports the automotive profile or not.

pinealservo, 2016-07-26:
The part you have referenced is not a complete solution; it's just a switch IC that has the hardware necessary to support AVB/TSN bridging. The behaviors you've asked about are all things that would be implemented in a software stack running on a separate host processor connected via xMII and SPI to the switch IC. Apparently NXP developed it in conjunction with TTTech; you can get an evaluation switch using that IC from them: https://www.tttech.com/products/automotive/automotive-ethernet/switches/de-switch-hermes/

Broadcom makes the BCM89200 and similar ICs; these have embedded host processors and some embedded PHYs as well, but your business would probably have to have a relationship with Broadcom to get any further information regarding the part, software, development boards, or much of anything.

Fortunately, gPTP works pretty much the same between two endpoints as it does between an endpoint and a switch, so you can do most endpoint gPTP validation without actually needing a switch.

DMats, 2016-08-06:
Ah, I see what you're saying.  Thank you, I believe that answers my question.

closed 2016-08-06 by DMats