Avnu/OpenAvnu #435: Multiple talkers on same computer

jpolin, 2016-07-19:
## Goal ##

On a single PC with 1 Intel i210, have:

- Talker A talking to Stream A (Traffic Class A)
- Talker B talking to Stream B (Traffic Class B)

## Approach ##

My proposed plan is to:

- Have single instances of the gPTP and MRP daemons
- Re-purpose simple_talker.c to talk on both streams. That is, write a single application that writes data for both classes to the same device. This way, there will only be once process talking to the MRP daemon and the igb device.

## Question ##

Do I need to modify any files other than simple_talker.c? Is it sufficient to make sure that the packet headers (Ethernet and 1722) contain the correct stream id/destination MAC/priority value, or will I need to modify code in the igb client (lib/igb/igb.c) or the igb driver (kmod/igb/igb_main.c) to have separate buffers for each stream/traffic class?

I'm worried about the following scenario:

I reserve 1 packet-per-interval for both streams. If 'A' is a packet for stream A and 'B' is a packet for stream B, and I call:

```c
igb_xmit(&igb_dev, 0, A);
igb_xmit(&igb_dev, 0, A);
igb_xmit(&igb_dev, 0, A);
igb_xmit(&igb_dev, 0, A);
igb_xmit(&igb_dev, 0, A);
igb_xmit(&igb_dev, 0, B);
```
then will the 'B' packet need to wait for 5 class-A intervals to pass before it will be sent (even though it can be sent at almost the same time)? It seems that the packets are stored in transmit rings on the NIC in the order that they're passed to the driver by ```igb_xmit(...)```, and only the "next up" packet can be sent at any given time. Thus, my 'B' packet will be blocked unnecessarily. This behavior is undesirable.

Any feedback on my approach or questions is more than welcome. I will continue experimenting with my system and provide updates as appropriate. Thanks for your help!

pinealservo, 2016-07-21:
The middle parameter for `gib_xmit` is the transmit queue to put the packet in.  If you're using both AVB traffic classes, they should really be in separate queues; the traffic shapers for the different traffic classes as configured by SRP operate on a per-queue basis.

Based on a quick look at the igb library, transmit queues 0 and 1 are mapped to class A and class B, respectively.

You may also want to look at the avtp_pipeline library, which has a lot of code for setting up and sending streams.  It's a rather large and complex library in comparison with simple_talker, but it's a port of production-level code and handles a lot of things simple_talker doesn't.

jpolin, 2016-07-21:
pinealservo, thanks for the suggestions. I'll play around with that middle argument, and take a look at the avtp_pipeline some more. 

Based on what you said, as well as the options given by functions such as ```igb_set_class_bandwidth(class_a_packets, class_b_packets, class_a_size, class_b_size)```, it seems that credit shaping on the talker is done on a class-by-class basis as opposed to a stream-by-stream basis. Adapting the example from above, I am worried about the scenario where:

- I am talking over 2 streams (Stream A1 and Stream A2) which are both class A
- I reserve 1 packet-per-interval for each stream
- I call:
```
// Send in single interval
igb_xmit(&igb_dev, 0, A1);
igb_xmit(&igb_dev, 0, A1);
igb_xmit(&igb_dev, 0, A1);
igb_xmit(&igb_dev, 0, A1);
igb_xmit(&igb_dev, 0, A2);
```

In this case, A2 will not be able to send right away even though it hasn't used its bandwidth for that interval yet. To address this scenario, do we need a stream-by-stream credit shaper in software (with queues for each stream) that controls the order of packets prior to calling ```igb_xmit(...)```?

pinealservo, 2016-07-21:
I don't know whether IGB will re-order its transmit queue based on launch time in the descriptors, but it would be better to not rely on that behavior even if it is supported.

So yes, you generally have to manage the order you fill each queue yourself; you'll want to walk through each stream in the queue's traffic class and add only the packets-per-interval for that stream before moving to the next.

I think avtp_pipeline takes care of this, but beware that it currently doesn't use the launch time functionality and relies on having the OS scheduler being fairly precise and consistent in low-latency scheduling, so it may not be suitable for Class A streams at present.  I'm working with the original contributors of the code to get programmed launch time functionality added, but there may still be issues when launch time is combined with multiple streams.

intel-ethernet, 2016-07-21:
The igb library doesn’t make any attempt to re-order frames, and frames are transmitted in the order they are posted to the HW queues.

So – to answer your question – with the I210  - you have to implement a sw module to sequence the packets from multiple streams onto a single outbound HW queue.

jpolin, 2016-07-28:
Thanks for the information pinealservo and intel-ethernet, this will prove helpful for our team going forward. If we develop a straightforward software layer with per-stream queues, I'll be sure to share it.

closed 2016-07-28 by andrew-elder