Avnu/OpenAvnu #436: gPTP: timestamping failures on link up/down events

merrilldav, 2016-07-20:
While running gPTPd on Linux from the master branch (compiled with generic architecture), we are seeing timestamping errors after a network port is reset using 'ifconfig ethx down' and 'ifconfig ethx up'.  We think we found a fix, but it is not covering an important edge case, as I'll describe below.

The initial error we saw:
`ERROR at 731 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72
Invalid TX`

gPTP never recovers and remains in master mode.  After killing the gPTPd daemon and restarting, it functions again.  This appears to relate to issue #254, which added fa7a482 to detect link state changes and call [timestamper_init()](https://github.com/AVnu/Open-AVB/blob/32dab06dd89427114d26be995ffdbf45289adb4e/daemons/gptp/common/ieee1588port.cpp#L588-L591) when LINKUP occurs.  However, in our testing we found that this change did not address our problem.

After some investigation, we discovered that the root cause is that the RX listener thread exits when it detects a failure receiving a packet, ensuring that no further packets will be received.  Removing the 'break' statement in the following code appears to fix the problem: [openPort()](https://github.com/AVnu/Open-AVB/blob/32dab06dd89427114d26be995ffdbf45289adb4e/daemons/gptp/common/ieee1588port.cpp#L354-L357)

However, we further found that the call to timestamper_init() causes several problems:
  1) an apparent deadlock shortly after LINKUP
  2) a ~40 byte memory leak each time the network cable is unplugged, seen in Valgrind
  3) adding redundant copies of the LinuxNetworkInterface object to iface_list in [HWTimestamper_init()](https://github.com/AVnu/Open-AVB/blob/32dab06dd89427114d26be995ffdbf45289adb4e/daemons/gptp/linux/src/linux_hal_generic.cpp#L252-L255)

We removed the call to timestamper_init(), and this appeared to make everything work.  However, after further testing running ifconfig down/up in a loop on the slave, we found that the _master_ would sometimes lose the ability to timestamp any TX packets:

`ERROR at 729 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72
ERROR at 1125 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response (Retrying-0), error=-72
ERROR at 1137 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response, error=-72
Invalid TX`

The master side was detecting the LINKDOWN and LINKUP events when ifconfig down/up was ran on the slave.  We have encountered this error both _with_ the call to timestamper_init in the LINKUP event, and without.  It reliably reproduces on the master within a few hours of running ifconfig down/up on the slave roughly twice a minute.  Additionally, killing the gPTPd daemon and restarting does NOT fix the problem on the master - only removing and re-adding the open_avb igb driver (rmmod / modprobe) or rebooting works.  Note that in these tests, the computers were directly connected (no switch in between).

It would seem that some sort of reset / re-initialization is needed for timestamping on LINKUP, but it needs to take into account the problems mentioned above.

kencarlino, 2016-07-20:
@merrilldav You may want to try this on the open-avb-next branch. There have been numerous changes to gPTP on this branch over the last couple of months that have not yet been merged to Master yet. There is one additional pull request that is still pending as well with some fixes to gPTP.

pinealservo, 2016-07-21:
I've seen some of these issues in the last few days on the open-avb-next branch, although I don't yet have @kencarlino's most recently merged patch (or the outstanding one, of course) applied yet.  I'll give some further feedback as I try out the new fixes.

merrilldav, 2016-07-21:
@kencarlino thanks for the feedback.  I reran my tests as described using the open-avb-next branch, including the igb driver.  The results were the same: the listener thread exited unless I removed the break statement in openPort() and it deadlocked unless I removed the call to timestamper_init() on LINKUP.  I made those changes on the slave and tested overnight with a script that calls ifconfig down / up in a loop.  As before, the master eventually fell into a state where it could not timestamp TX packets (note that I did not change the code on the master, so it still calls timestamper_init() on LINKUP).

There was one difference this time.  At some point in the night, it failed to initialize hardware timestamping at all:
`findPTPIndex: failed to open socket
Failed to find PTP device index
ERROR    : GPTP [07:13:44:863] Failed to initialize hardware timestamper, falling back to software timestamping
EXCEPTION: GPTP [07:13:45:608] PDelay Response Receipt Timeout
EXCEPTION: GPTP [07:14:01:612] LINK DOWN (maybe)`

After that, there were only LINK DOWN exceptions, as seen on the last line.

Note that I haven't tried the code in the outstanding pull request, yet.

merrilldav, 2016-07-21:
Confirming that I tried it with pull request #434, and the issue reproduced exactly as described.

davidcemin, 2016-07-21:
Its been awhile, but I tested this when I merged the automotive implementation over next, which was probably on this pull request:
https://github.com/AVnu/Open-AVB/pull/363

pinealservo, 2016-07-26:
There's definitely still work to be done on this. As of the latest code in open_avb_next, at least the following issues are still there:

LinuxTimestamperGeneric::HWTimestamper_init() is not designed to be called more than once, and doing so causes multiple problems:

1. It will create a new LinuxTimestamperGenericPrivate object each time and lose the old one. This has a cross_stamp_lock mutex in it, but it doesn't seem to be used anywhere. It also stores the PTP clock id/index.
2. It also pushes the interface passed in onto the LinuxTimestamperGeneric object's iface_list every time as noted by @merrilldav.
3. It opens /dev/ptpXX (with XX replaced, of course) again each time, but there's never a call to close that file descriptor. I don't think re-opening this is necessary; there's no code attached to ptp_open in the kernel PTP subsystem. I'm guessing that the failure @merrilldav saw in the index lookup was just simple file descriptor exhaustion due to calling open
4. In the call to findPhcIndex(), we can also leak socket descriptors if the ioctl to look up the PTP device index from its name fails. This probably never happens, but its a resource leak in a potentially live code path anyway.

In IEEE1588Port::timestamper_init(), we lose the whole HWTimestamper object if the init fails; it's not "leaked" in the sense that there's a reference to it on the stack frame for the main() function, but there's no way to get it back associated with the port. If there's no chance of recovery, it should be deleted, which will clear up its child objects as well. If we're going to continue calling it more than once, and there's a possibility of transient failure of the init, we should not NULL the reference if initialization succeeded in the past.

Problems aside, I don't think calling timestamper_init does any good relative to the eventual tx timestamping failures we've seen.

The only thing HWTimestamper_init() does that could be useful in a link_up situation is calling resetFrequencyAdjustment(), which just zeroes out the frequency adjustment parameter to LinuxTimestamperGeneric::Adjust( void *tmx ). This doesn't look like it would fix the issue with Tx timestamping, though; that stuff happens through a completely different interface, and it's all set up in LinuxTimestamperGeneric::post_init(). This is called automatically when the interface is built, and it sets the parameters on the socket necessary to talk to the kernel hardware timestamping subsystem. If calling these on a link change would fix things, I'd assume we'd be able to fix it by restarting gPTP rather than requiring an entire igb driver reload.

My recommendation is to remove the call to IEEE1588Port::timestamper_init() that's in the link up handler. If it's desired to reset the PTP clock (I don't think it is simply for the reason that in the absence of master clock info, we should free-run at the current clock rate rather than do an abrupt adjustment back to our base clock rate to avoid sudden frequency changes) we should have a timestamper_reset() method that does just that.

We should also clean up any possible resource leak issues that occur in error-exit cases along the init path even when the path is only called once; this would include adding descriptor closing calls to destructors where they don't currently exist.

Fixing the TX timestamping problem will require a bit more work; there's probably something going wrong with the interface between the kernel hardware timestamping code and the network device driver relating to link state handling, but I'm not very familiar with the innards of either of those things so it could be elsewhere as well.

pinealservo, 2016-09-23:
@merrildav Have you had a chance to see if #469 helped at all with the behavior you were seeing? Or have you made any further testing/observations about it?

iamdharm20, 2017-03-29:
Hi ,

i am facing this problem from log .
do anyone have solution for this.

root@KOR1064342VM:~/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build# /home/dus5cob/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build/obj/daemon_cl eth1 > ptp.log &
[4] 13897
root@KOR1064342VM:~/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build# INFO     : GPTP [17:42:25:659] gPTP starting
ERROR    : GPTP [17:42:25:660] Failed to configure timestamping: Operation not supported
ERROR    : GPTP [17:42:25:660] post_init failed

ERROR    : GPTP [17:42:25:660] failed to initialize port
^C
[4]-  Exit 255                /home/dus5cob/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build/obj/daemon_cl eth1 > ptp.log
root@KOR1064342VM:~/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build# 

Regards,
Dharmender sharma

andrew-elder, 2017-03-29:
My guess would be that either you are attempting to use the wrong interface, or the wrong IGB driver is loaded.

iamdharm20, 2017-03-29:
i am not sure but i am following this procedure : i have clone this branch :OpenAvnu-open-avb-next 
then i builded using make EXTRA_CFLAGS="-DIGB_PTP".
then i have loaded all these module.
sudo modprobe i2c_algo_bit
sudo modprobe dca
sudo modprobe ptp
sudo insmod /home/dus5cob/gitweb/OpenAvnu-open-avb-next/kmod/igb/igb_avb.ko

now if i run this gptp daemons i face this error : ERROR : GPTP [17:42:25:660] Failed to configure timestamping: Operation not supported

/home/dus5cob/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build/obj/daemon_cl eth1 > ptp.log &