Avnu/OpenAvnu #437: igb_avb: Support for reception time stamps on i210?

ginkgoz, 2016-07-21:
Dear all,

maybe a little off-topic, as it does not concern Open-AVB directly, but the igb_avb driver:
Is there a possibility to receive packets from the i210 including a reception time stamp using the igb_avb driver and the libigb included in the Open-AVB package here on GitHub? 

I only found functions for packet transmission by now and am not sure whether packet reception (including the reception timestamp) is possible this way.

Thanks in advance for any hints!

intel-ethernet, 2016-07-21:
Not yet, but there’s a patch coming which adds user-mode receive to libigb and the igb_avb driver …

If you want to simply get timestamps, the igb_avb driver has the PTP functions already to pass timestamps up the standard Linux network stack. This is what the PTP daemon uses for example.

This answer the question?

ekm

pinealservo, 2016-07-21:
If I remember correctly from my digging into NIC data sheets a while back, the receive timestamping is limited to PTP/gPTP packets.  Some NICs that support hardware timestamping just stick the receive timestamp at some location in the receive descriptors, so you can get timestamps for all received frames, but as far as I'm aware the Intel ones don't work that way.

Since you only get PTP timestamps anyway, it's best to just use the plumbing already built into the kernel unless you're doing something really out-of-the-ordinary.

intel-ethernet, 2016-07-21:
Some Intel adapters support timestamping all packets, and the I210 is capable of timestamping all packets, or timestamping all packets to a specific queue, or only “PTP” packets.

With the I210, the rx timestamp itself is stored in the packet data buffer (at the beginning of the frame) – so if you enable timestamps and are managing the frames directly, you have to adjust the start of packet past the timestamp data. Other Intel adapters store the rx timestamp at the *end* of the frame instead.

pinealservo, 2016-07-21:
Thanks for the correction!  Either I read things incorrectly, or I'm confusing the i210 with something else. Sorry for the confusion.

ginkgoz, 2016-07-22:
Thanks a lot for the great hints! 
I am sorry, I am still not sure what exactly to do on code level. Could you tell me what "managing the frames directly" means? Is there possibly some minimum code (or pseudo code) example? 
Thanks again for your help!

eric-mann, 2016-07-22:
The drivers IMO 'manage the frames directly'. The libigb library which OpenAVB uses also 'manages the frames directly' - it manages the descriptors of the device, and presents a 'packet API' to the upper level. So an app using the library doesn't know anything about the details how the packet is transfered between hw/sw.

What you are describing however may (or may not) require edits to the libigb (if you want to do this in userspace) .. or you can just figure out the right combination of system calls to enable timestamps on the interface, and use CMESGs to get the timestamps from the kernel. Without knowing exactly what level you are trying to work at (using libigb or std sockets), or what packets you want to timestamp (AVB / TSN ClassA/B traffic?) I can't be more specific on code recommendations.

ginkgoz, 2016-07-25:
Thanks a lot for the help!

Actually, I am trying to get a timestamp for any raw Ethernet packet of a certain EtherType that was received from a packet source in order to calculate packet timing statistics such as drift, jitter, etc. I am fully open whether to use libigb or standard raw Ethernet sockets -- whichever is easier to use according to your recommendation. A solution in user space would be great though.

Again, thanks a lot for your help!

eric-mann, 2016-07-25:
probably the easiest way is to re-use some of the code and modify some of the code from the example gPTP daemon.

Specifically, change the filtering from 'PTP_V2_EVENT' to 'ALL':

A)	Use standard socket() calls to enable timestamping of all packets. You could run the gPTP daemon and modify the function which enables timestamps on PTP packets to instead filter ‘ALL’. 

https://github.com/AVnu/Open-AVB/blob/master/daemons/gptp/linux/src/linux_hal_generic.cpp: LinuxTimestamperGeneric::post_init()
       device.ifr_data = (char *) &hwconfig;
       memset( &hwconfig, 0, sizeof( hwconfig ));
       **hwconfig.rx_filter = HWTSTAMP_FILTER_PTP_V2_EVENT;**
       hwconfig.tx_type = HWTSTAMP_TX_ON;
       err = ioctl( sd, SIOCSHWTSTAMP, &device );

You can also look at the code in the gPTP daemon (in the same file) which handles the CMESGs which return the HW timestamps. 

Lastly you create a raw socket on linux and bind it to the ethertype of your choice.

Hope this helps.

ginkgoz, 2016-07-26:
Thanks again! It seems to work, at least to some extent.
I have the impression that still either only SW timestamps are being used (although I read back all configuration to verify), or there is some other source of inaccuracy. When receiving hardware-generated cyclic packets with less than 100 nanoseconds jitter on the interarrival time, the reported reception timestamps (SO_TIMESTAMPNS CMESGs) have a jitter of some 200 microseconds. Are these really raw hardware timestamps of the i210, or might there be some issue e.g. when matching the hardware time stamps to the system time?

intel-ethernet, 2016-07-26:
Good question … I’d personally throw a few printk’s into interesting places in the igb driver and recompile … from your description it does sound like sw-based timestamps + the effects of driver interrupt moderation which typically adds 125 -> 250 usec latencies … which are on the same order as what you are seeing ..

The other quick test is to disable interrupt moderation with ethtool setting ‘rx_usecs’ to 0 and see if it changes the jitter/delays … if you are using HW timestamps, it shouldn’t matter what the interrupt moderation settings are for the packet timestamps.

ekm

ginkgoz, 2016-07-26:
You're right. When setting setting ‘rx_usecs’ to 0, the jitter is in the order of 10microseconds.

eric-mann, 2016-07-26:
Then you are getting SW-based timestamps, not HW-timestamps.

ginkgoz, 2016-07-26:
Found the problem. There was a bug in the binding of the socket to the network adapter, and I configured one adapter but listened via the socket to the other one.

Thank you all very much! It is amazing how people are volunteering to help others here!

closed 2016-07-26 by ginkgoz