Avnu/OpenAvnu #438: gPTP daemon attach to the wrong bus

rhii, 2016-07-22:
I have two I210 on my system. Using ethtool, eth0 is sitting on bus 0000:01:00.0 and link is up; while eth1 is sitting on bus 0000:02:00.0 and link is down.

I try to start the PPS on eth0 through -P switch and found that the actual PPS comes from eth1 instead of eth0.

Below are the logs:

root@intel-corei7-64:~# **ethtool eth0**
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        _Link detected: yes_
root@intel-corei7-64:~# **ethtool -i eth0**
driver: igb_avb
version: 5.3.2_AVB
firmware-version: 3.25, 0x8000060b, 1.654.0
expansion-rom-version:
_bus-info: 0000:01:00.0_
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no
root@intel-corei7-64:~# **ethtool eth1**
Settings for eth1:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        _Link detected: no_
root@intel-corei7-64:~# **ethtool -i eth1**
driver: igb_avb
version: 5.3.2_AVB
firmware-version:  0. 6-5
expansion-rom-version:
_bus-info: 0000:02:00.0_
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no
root@intel-corei7-64:~# **daemon_cl eth0 -P**
Using clock device: /dev/ptp0
_attaching to 0000:02:00.0_
[ 1034.213575] bind to iface 0000:02:00.0
[ 1034.221678] checking against adapter name 0000:01:00.0
[ 1034.230895] checking against adapter name 0000:02:00.0
Starting PDelay
*** Announce Timeout Expired - Becoming Master
New Grandmaster "A0:36:9F:FF:FE:68:49:5B" (previous "00:00:00:00:00:00:00:00")

Does anybody see this issue before? 

Regards,
Roland Hii

pinealservo, 2016-07-26:
I inadvertently discovered the reason for this when investigating #436.

As far as I can tell, none of the components of Open-AVB that directly use the igb library are going to work correctly in the presence of multiple igb interfaces.  They all use the same pci_connect() function, which walks through a PCI enumeration calling igb_probe() on each PCI device until one matches.  The interface you pass via the command line is only used for the APIs that go through standard kernel code. The gPTP daemon happens to do everything via standard kernel code *except* the PPS generation, and we apparently don't currently have any way of checking whether pci_connect() picked the interface we expected.

I don't currently have any idea when this might get fixed, so I would recommend only using a single igb interface if possible, or switching to the one that pci_connect() happens to pick for the PPS generation if you can't just have one igb interface installed.

eric-mann, 2016-07-26:
I had forgotten about this issue .. the implementations I am aware of which do use multiple I210s either hard-code the paths to the I210s and call the BIND ioctl directly - skipping the whole PCI enumeration bit, or they write some clever code which finds the ethernet interface name under sysfs and map it back to the PCI bus:device:funcion which the BIND ioctl requires.

e.g. if you want to connect to 'eth0', search under '/sys/devices' for it, like this example:

/sys/devices/pci0000:00/0000:00:1c.0/0000:09:00.0/net/eth0

They may be another method I'm not aware of as well ...