Avnu/OpenAvnu #448: ISO C++98 does not support the ‘hh’ gnu_printf length modifier [-Werror=format=] add("%02hhx ", byteMemory[currentPos + p]);

amustafa, 2016-08-05:
When running:

    mkdir tmp
    cd tmp
    cmake .. -G "Unix Makefiles"
    make

as instructed on the main page, I get:

    [ 11%] Building CXX object thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/MemoryLeakDetector.cpp.o
    /home/amustafa/workspace/Open-AVB/thirdparty/cpputest/src/CppUTest/MemoryLeakDetector.cpp: In member function ‘void SimpleStringBuffer::addMemoryDump(const void*, size_t)’:
    /home/amustafa/workspace/Open-AVB/thirdparty/cpputest/src/CppUTest/MemoryLeakDetector.cpp:75:54: error: ISO C++98 does not support the ‘hh’ gnu_printf length modifier [-Werror=format=]
             add("%02hhx ", byteMemory[currentPos + p]);
                                                      ^
    cc1plus: all warnings being treated as errors
    thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/build.make:326: recipe for target  'thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/MemoryLeakDetector.cpp.o' failed
    make[2]: *** [thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/MemoryLeakDetector.cpp.o] Error 1
    CMakeFiles/Makefile2:1033: recipe for target 'thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/all' failed
    make[1]: *** [thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/all] Error 2
    Makefile:138: recipe for target 'all' failed
    make: *** [all] Error 2

I was able to get around this by opening the tmp/CMaleCache.txt file and changing the following flag:

       //Compile with C++11 support
      -  C++11:BOOL=OFF
      +  C++11:BOOL=ON

t-hofmann, 2016-08-25:
simple change in
Open-AVB/thirdparty/cpputest/src/CppUTest/MemoryLeakDetector.cpp
line 75
from
         add("%02hhx ", byteMemory[currentPos + p]);
to
         add("%02hx ", byteMemory[currentPos + p]);

This same change is done in the upstream repository "cpputest", unfortunately cloning from that original repository lead (in my case) to other problems. So I would recommend the above change.

andrew-elder, 2016-08-25:
Thanks for the update @t-hofmann. Open-AVB currently uses cpputest as a submodule that points to the cpputest master. We would have to make our own fork and then patch and use that to do what you suggest. Is there no tag in the parent repo that works ok? What were the other problems you encounter when you updated to the latest upstream version?

pinealservo, 2016-08-25:
Maybe we can set that setting from the command line or environment as the build process for cpputest is invoked?

Some potential solutions in the order of desirability (to me, anyway):

1. Find a ref in upstream cpputest that works for everyone and change our submodule ref to that.
2. Change the setting via environment or command line flag
3. Apply a small patch to MemoryLeakDetector.cpp before building
4. Our own fork of cpputest

I really don't like 3 or 4, so hopefully we can make 1 or at least 2 work.

pinealservo, 2016-09-13:
I have a potential solution; try applying this change to the toplevel CMakeLists.txt:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index cba5d83..f6cc077 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,6 +1,8 @@
 cmake_minimum_required (VERSION 2.8) 
 project (open-avb)
 enable_testing()
+
+set(C++11 ON CACHE BOOL "Compile with C++11 support" FORCE)
 add_subdirectory("thirdparty/cpputest")
 add_subdirectory("daemons/common/tests")
 add_subdirectory("daemons/mrpd")
```
I didn't get the error before, so I'm not sure if this has fixed it.  I have a local branch I could make a pull request from (it also fixes some warnings that crop up in the tests as a result), but it'd be nice to know if it's going to help or cause other problems first.

@t-hofmann and @amustafa -- could you try this out and see if it fixes your issue?

bdthomsen, 2016-09-15:
I tried the toplevel CMakeLists.txt fix on Linux Mint 18 Cinnamon 64-bit, which exhibits the compile error.  With the change, the compile issue was resolved, and I didn't see any other issues as a result of it.

andrew-elder, 2016-09-15:
Great!

amustafa, 2016-09-21:
Confirmed, @pinealservo solution avoided the error.

closed 2016-09-23 by pinealservo