Avnu/OpenAvnu #449: Is it necessary to build gPTP with ARCH=I210 for it to work on an Intel I210 card?

DMats, 2016-08-07:
Is #390 actually a problem for those of us using Intel I210 cards?
What is the effect of building gPTP without any ARCH variable set?

jfornal, 2016-08-12:
As you may notice in daemons/gptp/linux/build/Makefile there is an if statement dedicated for the ARCH variable. If it is set to I210, some I210 specific object files are generated and included in the daemon_cl output located in /obj folder (the files differ in size and partly the content - checked in Hex Editor - if you compare the ones with and without ARCH flag set). Moreover additional CFLAGS are included (WITH_IGBLIB) which are used to distinguish some functionalities in linux_hal_generic files.  So I would say that building that daemon without ARCH set, make it more generic.  Nevertheless all the tests I managed to perform previously included the ARCH=I210, so can't be sure whether they introduce any different behaviour of the code.

pinealservo, 2016-09-12:
When no ARCH is set, only the capabilities available through standardized interfaces are available. In the case of i210, most capabilities are available through the standard interfaces.  However, the PPS capability requires the arch-specific build.

I think the Linux kernel interfaces now have PPS functionality for drivers that support it (which the kernel igb driver appears to, while igb_avb appears not to), but I don't know how it relates to the PPS support in `linux_hal_i210.cpp`.

andrew-elder, 2016-09-22:
@DMats  - has your question been answered?

DMats, 2016-09-22:
Yes this answers my question.  Thank you Levi and jfornal

closed 2016-09-22 by andrew-elder