Avnu/OpenAvnu #456: gPTP doesn't correctly read PHY delay adjustments from .ini file

christopher-s-hall, 2016-08-12:
Changing these values in the .ini file has no effect. The output of gPTP indicates that the results were parsed correctly:

...
phy_delay_gb_tx: 7000
phy_delay_gb_rx: 7000
phy_delay_mb_tx: 7000
phy_delay_mb_rx: 7000
...

The configuration file appears to be read (in daemon_cl) after the port is already initialized.

andrew-elder, 2016-08-18:
Is anyone working on this?

andrew-elder, 2016-08-23:
@christopher-s-hall - I think this is fixed now, correct?

christopher-s-hall, 2016-08-23:
No. If it's still not working correctly, I'll take a look when I have some time. It will probably be weeks before I have time.

andrew-elder, 2016-08-23:
My bad. I thought this commit https://github.com/AVnu/Open-AVB/pull/450/commits/e884d818b9671fc4ffab0de800703e67463a056e fixed it....   Maybe I mis-understand your response? Are you saying "No it should no longer be a bug/issue"?  Or, maybe "No I haven't tested the fix"?

christopher-s-hall, 2016-08-23:
That pull request is closed out and contains other stuff as well. If you can “cherry pick” that out then go for it.

andrew-elder, 2016-08-24:
I would maintain that all the changes in the closed out pull request look worthwhile to me. With your permission, I'm happy to grab your changes and create new pull requests. Nothing more for you to do.

pinealservo, 2016-09-12:
Fixed in #462

closed 2016-09-12 by pinealservo