Avnu/OpenAvnu #457: CMake cache clean build rules?

pinealservo, 2016-08-18:
Had an issue building avtp_pipeline that ended up being due to a failed earlier build and cmake's caching of its build instruction metadata.

I tried the clean rule:

    make avtp_pipeline_clean

But it did not delete the cmake metadata, and so it continued to fail until I figured out which files were the cache and deleted them manually.

Do we want a normal "clean" to delete the cmake cache?  Do we want a separate build target for cleanup of cache files?

MarcinMiklas, 2016-08-19:
avtp_pipeline_clean cleans whole lib/avtp_pipeline/build directory where I though all cmake cache is kept (because cmake is run in this directory).
Which cmake cache files are kept outside of that dir? It would be ok to add cleaning those additional cmake cache files to avtp_pipeline_clean target.

pinealservo, 2016-08-21:
Somehow, in the situation I was debugging, the metadata had ended up in `lib/avtp_pipeline` rather than in `lib/avtp_pipeline/build`. After this occurred, running cmake from the build directory would detect the cached files one directory up rather than making new ones, and cleaning didn't fix the problem because the clean rule assumed that removing `build` and its contents was sufficient.

I guess there's not much that can be done about automating the fixing of this sort of build problem; we *could* have a rule in the top-level makefile that just recursively finds all instances of `CMakeCache.txt` and `CMakeFiles` in the project and remove each of them, but someone would have to know when it was appropriate to run it, and someone who knows that probably doesn't need it in the first place.

Thanks for looking into it and commenting, Marcin.

closed 2016-08-21 by pinealservo