Avnu/OpenAvnu #460: gPTP: toByteString() function has no size checking

andrew-elder, 2016-08-23:
Current implementation blindly copies to target pointer without checking the size of the passed in memory area. Ideally there would be a size_t sizeof_byte_str_array parameter passed in by the caller so that the size could be checked before memcpy() type operations are undertaken.

jfornal, 2016-09-06:
I just created a Pull request for that issue (#468) - please review the changes and check whether this is exactly what you thought should be introduced into the code

pinealservo, 2016-09-06:
See my comment in #468 where I suggest that modifying `toByteString` might not be the best way to avoid blind copying. It's only blind if you don't know there's room before calling. It's basically a `memcpy` that automatically picks the right value for its length argument; I believe it's meant to have the same calling contract as `memcpy` itself and I think a better resolution would be to make note of the contract in the documentation and review callers to ensure there aren't any cases where the contract might be violated.  I don't think there are at present, though there are some places where assertions might be added to ensure protocol changes or different network interface types don't introduce regressions.

andrew-elder, 2016-09-09:
I'm going to close this issue for the moment. There really is no bug. If someone wants to open a new issue where parameters to `toByteString` are checked, they can.

closed 2016-09-09 by andrew-elder