Avnu/OpenAvnu #465: gPTP - SYNC_INTERVAL_TIMEOUT_EXPIRES automotive sync interval seems incorrect

andrew-elder, 2016-09-01:
This code seems to be incorrect to me:

https://github.com/AVnu/Open-AVB/blob/open-avb-next/daemons/gptp/common/ieee1588port.cpp#L1099

The comment talks of 16 millisecond timer in the if (!automotive_profile) section of the code.

What do the automotive experts think?

andrew-elder, 2016-09-01:
Yes, but the accelerated logic seems to be selected by the *if (!automotive_profile)* code path - notice the  "!".

andrew-elder, 2016-09-01:
Does a wireshark trace show the accelerated sync path works?

andrew-elder, 2016-09-01:
Thanks for the comment @kencarlino.

christopher-s-hall, 2016-09-01:
The general idea was to fire off several (I think there was a parameter for this) sync messages in quick succession.  I may have changed the number to make the comment not match. The pdelay wasn't supposed to start (using cached values) until all of these "fast" syncs were sent.

I don't know of anyone using this code and it can probably be removed since it doesn't comply with any standards.  It is also true that the code implementing the automotive profile renders this code mostly useless.  I say dump it.

andrew-elder, 2016-09-02:
Thanks for the comments @christopher-s-hall  and @kencarlino. Just to be sure I understand what is being said here, I'm going to restate what I understand is being said and proposed code changes.

The main point I take from comments so far is that there is no longer a requirement to fire off some number of sync messages in quick succession. That means I can remove *accelerated_sync_count* from here
https://github.com/AVnu/Open-AVB/blob/0c8a455ec1d84ff09608dcd74cfe5726da59385a/daemons/gptp/common/avbts_port.hpp#L223
and *_accelerated_sync_count* from
https://github.com/AVnu/Open-AVB/blob/0c8a455ec1d84ff09608dcd74cfe5726da59385a/daemons/gptp/common/avbts_port.hpp#L322
and clean up any associated interfaces/references.

And then this if() else sequence 
https://github.com/AVnu/Open-AVB/blob/0c8a455ec1d84ff09608dcd74cfe5726da59385a/daemons/gptp/common/ieee1588port.cpp#L1099
collapses to
```
		syncDone();
		wait_time = ((long long) (pow((double)2, getSyncInterval()) * 1000000000.0));
		wait_time = wait_time > EVENT_TIMER_GRANULARITY ? wait_time : EVENT_TIMER_GRANULARITY;
		startSyncIntervalTimer(wait_time);
```
?

andrew-elder, 2016-09-07:
@christopher-s-hall - can you please take a quick look at my previous post on this issue/thread and confirm whether or not I'm on the correct track. I'm not 100% sure I correctly understood your suggestion.

christopher-s-hall, 2016-09-08:
This looks fine to me. I think there might be a command line argument (or two) to remove as well.