Avnu/OpenAvnu #467: gptp daemon: Link delay 35184372096401 beyond neighborPropDelayThresh; not AsCapable

Fernando-Sanchez, 2016-09-05:
Hi,

I am using  Ubuntu 14.04.2 LTS and an Ethernet controller: Intel Corporation Ethernet Connection I217-LM (rev 04). Initially I was using the gptp daemon with the default gptp_cfg.ini file, and the result was as follows:

demo@sa2:~/Downloads/Open-AVB-master/daemons/gptp$ sudo ./linux/build/obj/daemon_cl eth1 -F gptp_cfg.ini
INFO     : GPTP [11:19:31:059] gPTP starting
Using clock device: /dev/ptp0
priority1 = 248
announceReceiptTimeout: 3
syncReceiptTimeout: 3
phy_delay_gb_tx: 184
phy_delay_gb_rx: 382
phy_delay_mb_tx: 1044
phy_delay_mb_rx: 2133
neighborPropDelayThresh: 800
syncReceiptThreshold: 8
STATUS   : GPTP [11:19:31:061] Starting PDelay
ERROR    : GPTP [11:19:31:095] Link delay 7656 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:19:31:951] Switching to Slave
STATUS   : GPTP [11:19:31:951] New Grandmaster "88:1D:FC:FF:FE:FD:E8:00" (previous "00:00:00:00:00:00:00:00")
ERROR    : GPTP [11:19:32:109] Link delay 7700 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [11:19:33:127] Link delay 7588 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [11:19:34:124] Link delay 7344 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [11:19:35:135] Link delay 7632 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [11:19:36:147] Link delay 7220 beyond neighborPropDelayThresh; not AsCapable

So I modified the file gptp_cfg.ini, to set neighborPropDelayThresh to 8000. But that did not solve the issue, now I just see bigger link delays:

demo@sa2:~/Downloads/Open-AVB-master/daemons/gptp$ sudo ./linux/build/obj/daemon_cl eth1 -F gptp_cfg.ini
INFO     : GPTP [11:23:53:101] gPTP starting
Using clock device: /dev/ptp0
priority1 = 248
announceReceiptTimeout: 3
syncReceiptTimeout: 3
phy_delay_gb_tx: 184
phy_delay_gb_rx: 382
phy_delay_mb_tx: 1044
phy_delay_mb_rx: 2133
neighborPropDelayThresh: 8000
syncReceiptThreshold: 8
STATUS   : GPTP [11:23:53:103] Starting PDelay
STATUS   : GPTP [11:23:53:137] AsCapable: Enabled
STATUS   : GPTP [11:23:53:562] Switching to Slave
STATUS   : GPTP [11:23:53:562] New Grandmaster "88:1D:FC:FF:FE:FD:E8:00" (previous "00:00:00:00:00:00:00:00")
ERROR    : GPTP [11:25:18:882] Link delay 35184372096053 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:25:18:882] AsCapable: Disabled
STATUS   : GPTP [11:25:19:894] AsCapable: Enabled
ERROR    : GPTP [11:25:35:031] Link delay 35184372096528 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:25:35:031] AsCapable: Disabled
STATUS   : GPTP [11:25:36:043] AsCapable: Enabled
ERROR    : GPTP [11:27:49:240] Link delay 35184372096121 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:27:49:240] AsCapable: Disabled
STATUS   : GPTP [11:27:50:252] AsCapable: Enabled
ERROR    : GPTP [11:27:53:277] Link delay 35184372096409 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:27:53:277] AsCapable: Disabled
STATUS   : GPTP [11:27:54:288] AsCapable: Enabled
ERROR    : GPTP [11:28:09:426] Link delay 35184372096537 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:28:09:426] AsCapable: Disabled
STATUS   : GPTP [11:28:10:438] AsCapable: Enabled
ERROR    : GPTP [11:28:13:464] Link delay 35184372096445 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:28:13:464] AsCapable: Disabled
STATUS   : GPTP [11:28:14:475] AsCapable: Enabled
ERROR    : GPTP [11:28:33:650] Link delay 35184372096102 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [11:28:33:650] AsCapable: Disabled
STATUS   : GPTP [11:28:34:663] AsCapable: Enabled
ERROR    : GPTP [11:30:04:434] Link delay 35184372096100 beyond neighborPropDelayThresh; not AsCapable

I don´t know whether it is a bug in the code or I am missing something? Your help is appreciated. Thanks!

pinealservo, 2016-09-06:
The newer link delays are pretty strange... they'd be in the same range as the previous ones except they've got bit 45 set. Something funny definitely must have gone wrong in the math, but there are probably some things that are wrong with the situation that led to the math issue, because the code is fairly well tested in the zone of operation it's designed to work in.

8000ns propagation delay is well outside of the reasonable range for gPTP operation. The signal propagation time for a 100m UTP cable is about 5000ns, so you couldn't possibly get a legitimate ethernet link delay of 8000ns. You must have some sort of signal repeater (i.e. a really slow media converter, a switch that doesn't do gPTP, etc.) between the two that's increasing the delay, and the whole point of the AsCapable determination is to disallow that.

If it's genuinely measuring a point-to-point link delay of ~7500ns or so, something is not right with your setup. Possibly you're not really getting hardware timestamps? I haven't tried running OpenAVB gPTP on anything other than an i210, though there are definitely other NICs that should work.

If you can get the link delay down to a reasonable level, you can try tweaking the PHY delay settings; they're set up for the i210 by default and the actual values for the i217 will no doubt be different. If you're going to adjust the PHY delay values via the INI file, you'll need to switch to the open-avb-next branch, because a bug in the order of reading the INI file and initializing some things got fixed recently. Specifying them on the command line should work in either branch.

You might also look at what the device that's acting as GM thinks the link delay is.

Fernando-Sanchez, 2016-09-06:
The link delay was also huge from the device acting as GM point of view.

I am using an I210 network interface now, and the link delay is about 35ns now, and everything works fine.

closed 2016-09-06 by Fernando-Sanchez

eric-mann, 2016-09-06:
I am not aware of a detailed engineering characterization of the I217 timestamp behavior. From what I know about the internal architecture of the device, it wouldn't surprise me to see 'very large' seeming path delays on timestamps. The driver would need to compensate the timestamps internally before presenting to the OS or application (or fixup the delays in the INI file).

shirArg, 2018-12-02:
@Fernando-Sanchez did you ever managed to solve this?

shirArg, 2018-12-02:
Hi, not exactly the same problem, but- i'm using the gptp with two laptops, one with I218-LM running Ubuntu 18.04, and one with I219-LM running Windows 10. 

I got this while trying to use it- 

INFO     : GPTP [16:51:09:271] Starting IPC listener thread succeeded
Opening: rpcap://\Device\NPF_{33F88F37-D54C-4965-B2FA-EA134B078A74}
INFO     : GPTP [16:51:09:400] Adapter UID: {33F88F37-D54C-4965-B2FA-EA134B078A74}

STATUS   : GPTP [16:51:09:403] Starting PDelay
ERROR    : GPTP [16:51:09:538] *** Received an event packet but cannot retrieve timestamp, discarding. messageType=2,error=-72
ERROR    : GPTP [16:51:09:543] Discarding invalid message
ERROR    : GPTP [16:51:09:670] *** Received an event packet but cannot retrieve timestamp, discarding. messageType=3,error=-72
ERROR    : GPTP [16:51:09:675] Discarding invalid message
ERROR    : GPTP [16:51:09:676] >>> Received PDelay followup but no RESPONSE exists
STATUS   : GPTP [16:51:10:436] AsCapable: Enabled
ERROR    : GPTP [16:51:11:440] Link delay 3612 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [16:51:11:456] AsCapable: Disabled
STATUS   : GPTP [16:51:12:403] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [16:51:12:410] New Grandmaster "D4:81:D7:FF:FE:5D:F5:DB" (previous "00:00:00:00:00:00:00:00")
ERROR    : GPTP [16:51:12:440] Link delay **-1200** beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [16:51:13:440] AsCapable: Enabled
ERROR    : GPTP [16:51:14:440] Link delay 2858 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [16:51:14:449] AsCapable: Disabled
ERROR    : GPTP [16:51:15:442] Link delay 2940 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [16:51:16:442] Link delay 3143 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [16:51:17:443] AsCapable: Enabled
ERROR    : GPTP [16:51:18:443] Link delay 2887 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [16:51:18:454] AsCapable: Disabled
STATUS   : GPTP [16:51:19:443] AsCapable: Enabled
ERROR    : GPTP [16:51:20:445] Link delay 3460 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [16:51:20:459] AsCapable: Disabled
ERROR    : GPTP [16:51:21:445] Link delay 3466 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [16:51:22:446] Link delay 2950 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [16:51:23:445] Link delay 2981 beyond neighborPropDelayThresh; not AsCapable

This is after I changed the neighborPropDelayThresh in gptp_cfg.ini file to 10000 (and built everything again), in both computers. 

Any idea why the change didn't help? And what's going on with the negative delay?...

Thanks!

pinealservo, 2018-12-04:
Unfortunately there are a lot of potential problem spots and it's very difficult to debug remotely. I'll try to give you some pointers for places to look.

Correct delay calculation relies on correct PHY delay timing, and errors here can easily cause negative delay times to be reported. It's very odd that you get timestamp retrieval messages at first, but apparently it starts working soon afterward. Probably the delays are way off; on Windows they're set by default to "generic PCH delay" values--someone more knowledgeable about Intel hardware would have to explain what that means and whether they bear any resemblance to actual PHY delays for your NICs. I think you'll have to edit the code in windows/daemon_cl/daemon_cl.cpp to change the timing values; there are some constants for "generic PCH" and the i210 at the top of the file. You can specify these in Linux on the command line, and it might be one of the options that the config file will set too.

The configuration code in gPTP is currently a bit of a mess, and it's inconsistent between Windows and Linux.  I don't think the INI parser was ever completely integrated, and it doesn't seem to be integrated on Windows at all. It *should* work on Linux for setting the neighborPropDelayThresh value, though. It mostly works for the use cases that people using it have exercised, but beyond that it's got problems.

I think the only way to change that threshold in Windows right now is to edit common/common_port.hpp and change NEIGHBOR_PROP_DELAY_THRESH.

The annoying popups are apparently a new sanitizer feature in the Windows C++ runtime. If you hit "Retry" to debug and find the variable it's complaining about, you could add an initializer to it in the code and then send a pull request. :)

shirArg, 2018-12-04:
Regarding the "ret" problem- the problem is in common/ether_port.cpp - in the function EtherPort::_processEvent, 'ret' is used, but not treated in some of the switch cases (specially the ones that treat the errors). I initialized it to 'false', but i'm really clueless about whether it should be true or false, so I'll avoid sending a pull request for now :)

Regarding the NEIGHBOR_PROP_DELAY_THRESH - changing it in the common_port.hpp in both computers seems to work.

Thank you very much! :)

dancunogenda, 2021-06-09:
Did someone find a permanent solution to this problem. I am using TSN evaluation kits that support IEEE 802.1AS and IEEE 802.1AS-Rev but when I run this software on a PC (Grandmaster) and use the TSN kits (3 as slaves), I get a huge value for the delay which is a huge number compared to the 800 nanoseconds in the code. A solution or guide on how to solve this would be appreciated.