Avnu/OpenAvnu #472: Investigate Windows CI build via appveyor.com

pinealservo, 2016-09-07:
Since we set up the Travis CI build check hook, a Windows-based CI service that's also free for open source projects has become available from http://appveyor.com/ that we should investigate using to help make sure we keep the Windows build from breaking.

ahogen, 2017-06-20:
Looking at the [commit history](https://github.com/AVnu/OpenAvnu/commits/master), Appveyor has been setup, but not correctly. It always fails. 

The last one failed with the following error...
```
Build started
git clone -q --branch=master https://github.com/AVnu/OpenAvnu.git C:\projects\openavnu
git checkout -qF <hash here>
Specify a project or solution file. The directory does not contain a project or solution file..
```
Source: https://ci.appveyor.com/project/andrew-elder/openavnu/build/1.0.11

andrew-elder, 2017-06-20:
Appveyor won't work correctly for a while because there are chunks of code that do not yet have a Windows equivalent. That said, it would be better if there were only a few errors to address.

pinealservo, 2017-06-21:
Thanks for making that initial effort, @andrew-elder. Do you have a list of things that need to be addressed you could throw here in the off chance @ahogen or some other industrious bystander would like to take a crack at getting the build into a state that would make appveyor happy? Does it need a separate Cmake configuration or something like that?

andrew-elder, 2017-06-21:
I'm thinking perhaps I should create a feature_appveyor branch so that others can work on the issue(s). The one that stumped me was actually installing cmake 3.4 for travis. I figure whatever I add, travis should still work for Linux builds. I added gPTP build via cmake under travis and that triggered errors that I ran out of time to fix.

Aside from travis build of gPTP via cmake, there are 4 MSVC errors that need to be addressed. I think there are some Windows "chunks" missing in MAAP that prevented linking....

bdthomsen, 2017-06-21:
Yes, there are missing Windows "chunks" in MAAP.  While I got the CppUTests working for the Windows build, the main daemon code for Windows was never implemented, and maap_main.c is essentially empty.  It would be trivial to add an empty main() function so that the daemon compiles; adding the real support will be a much bigger effort.

andrew-elder, 2017-09-21:
See https://github.com/AVnu/OpenAvnu/commit/bffc91619ffff1990d81e7326d7d5879a705f523

The current challenge is getting travis to use a newer version of cmake that is required to build gPTP using cmake under windows. There is a feature-appveyor branch that anyone can contribute to.

andrew-elder, 2017-10-09:
See https://github.com/AVnu/OpenAvnu/pull/702 - this is done.

closed 2017-10-09 by andrew-elder