Avnu/OpenAvnu #473: Array subscript is above array bounds warning in maap_linux.c

rhii, 2016-09-08:
I am adding -O2 to Open AVB daemons Makefile CFLAGS and one of the warning shown is:

```
maap_linux.c: In function ‘main’:
maap_linux.c:198:16: warning: array subscript is above array bounds [-Warray-bounds]
  seed = src_mac[6] + time(NULL);
                ^
```

I have checked the source code daemons/maap/linux/maap_linux.c, in function main(), src_mac is declared as 6 bytes and seed is an unsigned long.
```
uint32_t seed;
uint8_t src_mac[6];
```

May I know if it is a mistake or it is by purpose to use src_mac[6] to assign the seed?

Regards,
Roland Hii

pinealservo, 2016-09-09:
That is probably a bug.  My guess is that they meant to add src_mac[5] in order to make the random seed not entirely dependent on the start time of the daemon. Even if it was intentional, it's undefined behavior and shouldn't be done that way.

The contributor of the maap code hasn't been active for a very long time, and the current state of it is very far from being a complete and correct implementation, but it may work for you as long as no one attempts to reserve more than a single multicast address at a time.

rhii, 2016-09-22:
The pull request has been merged. I suppose this issue can be closed now, or we need some one else to verify?

pinealservo, 2016-09-22:
Thanks for the fix!

PR #477 (2016-09-21): Fixes to Makefiles and compiler warnings and errors
Since added simple_rx, the clean target does not properly clean all the generated object and executable files. The changes here corrected the corresponding Makefiles to be able to remove all the generated files.

Besides that, when added -O2 to Open AVB daemons Makefile CFLAGS, there are some compile warnings and errors shown. The changes here also fix those warnings and errors.

This should fix #473 too.

Regards,
Roland Hii
closed 2016-09-22 by pinealservo