Avnu/OpenAvnu #474: igb_avb: Computer crashes after running simple_talker or avtp_pipeline talker twice.

MarcinMiklas, 2016-09-12:
See comments to pull request for more details
#454

pinealservo, 2016-09-12:
I was fairly sure I'd done my recent tests on low-powered machines with the referenced pull request included; I didn't get any kernel oops.  I'll re-run them, though, to be sure.

MarcinMiklas, 2016-09-12:
Talkers usually run without problems, for the first time. Stop them and start again and you will see the issue. Maybe some more careful cleanup is needed, I don't know. But computer crashes so badly that all I could do is to press power button for 5 seconds to make it turn off. It's funny because everything stays on the screen, but computer doesn't react to mouse, keyboard. Kernel crash is not displayed on the screen, I could not find anything related to crash in syslog.

pinealservo, 2016-09-12:
OK, I have reproduced the issue now, it's definitely a problem.

pinealservo, 2016-09-12:
@cirust20 and @intel-ethernet if you could take a look at this, I'll investigate further myself.  I was able to reproduce is with simple_talker and simple_listener on two back-to-back Linux PCs with i210 NICs.  The first attempt starts up fine, but stopping and re-starting simple_talker leads to a crash.

If you haven't run simple_talker or simple_listener in a while, they now talk to gPTP and SRP and need to confirm gPTP lock and SRP reservation before they start their streaming.

eric-mann, 2016-09-12:
By inspection, this line of code in igb_main.c:igb_open_file() looks odd:

igb_priv = kzalloc(sizeof(struct igb_private_data *), GFP_KERNEL);

wouldn't you want to allocate sizeof(struct igb_private_data) instead of a pointer ?

pinealservo, 2016-09-12:
The kernel logs are extensive; it doesn't crash immediately, so there are a series of lengthy oops dumps on the console. I'm guessing it's corrupting the kernel memory allocator's data structures, since the functions that show up in the logs vary and change from crash to crash.

pinealservo, 2016-09-12:
@eric-mann is the winner.  Allocating the size of the struct instead of the pointer seems to resolve the problem.

andrew-elder, 2016-09-12:
Good spot. Feel free to create and merge a pull request @pinealservo. Seems like the fix is obvious, so we don't need lots of code reviewers.

pinealservo, 2016-09-12:
Just submitted pull request #476

closed 2016-09-19 by pinealservo

cirust20, 2016-10-10:
sorry for late. I was on vacation.
thanks for fix.