Avnu/OpenAvnu #475: Incorrect Ethernet Header

dmlarson4, 2016-09-12:
Using the simple_talker and simple_rx programs in OpenAvb, we are not seeing the correct Ethernet header received. I print the headers in hexadecimal as they are sent and as they are received and see that the 802.1Q header (with TPID 0x8100 and the two bytes after) are not sent at all. This means the packet size in the listener is four bytes less than what is sent. 

Do you know why I am not seeing the right Ethernet header information?

Thanks,
dmlarson4

simple_talker output: (note bytes 12-15)
```
sent size: 98
[0]91
[1]e0
[2]f0
[3]00
[4]0e
[5]80
[6]a0
[7]36
[8]9f
[9]95
[10]5e
[11]ca
[12]81 <--
[13]00 <--
[14]00 <--
[15]00 <--
[16]22
[17]f0
[18]00
[19]81
[20]9d
[21]00
[22]a0
[23]36
[24]9f
[25]95
[26]5e
[27]ca
[28]00
[29]00
[30]73
[31]64
[32]9d
[33]4b
[34]00
[35]00
[36]00
[37]00
[38]00
[39]20
[40]5f
[41]a0
[42]3f
[43]01
[44]00
[45]5c
[46]90
[47]02
[48]ff
[49]ff
[50]40
[51]00
[52]00
[53]00
[54]40
[55]00
[56]00
[57]00
[58]40
[59]00
[60]00
[61]00
[62]40
[63]00
[64]00
[65]00
[66]40
[67]00
[68]00
[69]00
[70]40
[71]00
[72]00
[73]00
[74]40
[75]00
[76]00
[77]00
[78]40
[79]00
[80]00
[81]00
[82]40
[83]00
[84]00
[85]00
[86]40
[87]00
[88]00
[89]00
[90]40
[91]00
[92]00
[93]00
[94]40
[95]00
[96]00
[97]00
```
simple_rx output:
```
recieved size: 94
[0] 91
[1] e0
[2] f0
[3] 00
[4] 0e
[5] 80
[6] a0
[7] 36
[8] 9f
[9] 95
[10] 5e
[11] ca
[12] 22
[13] f0
[14] 00
[15] 81
[16] 25
[17] 00
[18] a0
[19] 36
[20] 9f
[21] 95
[22] 5e
[23] ca
[24] 00
[25] 00
[26] 6e
[27] af
[28] 2b
[29] 8b
[30] 00
[31] 00
[32] 00
[33] 00
[34] 00
[35] 20
[36] 5f
[37] a0
[38] 3f
[39] 01
[40] 00
[41] bc
[42] 90
[43] 02
[44] ff
[45] ff
[46] 40
[47] 00
[48] 00
[49] 00
[50] 40
[51] 00
[52] 00
[53] 00
[54] 40
[55] 00
[56] 00
[57] 00
[58] 40
[59] 00
[60] 00
[61] 00
[62] 40
[63] 00
[64] 00
[65] 00
[66] 40
[67] 00
[68] 00
[69] 00
[70] 40
[71] 00
[72] 00
[73] 00
[74] 40
[75] 00
[76] 00
[77] 00
[78] 40
[79] 00
[80] 00
[81] 00
[82] 40
[83] 00
[84] 00
[85] 00
[86] 40
[87] 00
[88] 00
[89] 00
[90] 40
[91] 00
[92] 00
[93] 00
```

eric-mann, 2016-09-12:
The receiver is (most probably) stripping the VLAN tags ... from the I210 datasheet:

7.4.3.2 Stripping 802.1Q Tags on Receives
Software might instruct the I210 to strip 802.1Q VLAN tags from received packets. If VLAN stripping is
enabled and the incoming packet is an 802.1Q VLAN packet (its Ethernet Type field matched the VET),
then the I210 strips the 4 byte VLAN tag from the packet, and stores the TCI in the VLAN Tag field (see
Figure 7-4 and See “Receive UDP Fragmentation Checksum) of the receive descriptor.
The I210 also sets the VP bit in the receive descriptor to indicate that the packet had a VLAN tag that
was stripped. If the CTRL.VME bit is not set, the 802.1Q packets can still be received if they pass the
receive filter, but the VLAN tag is not stripped and the VP bit is not set.
VLAN stripping can be enabled using two different modes:
1. By setting the DVMOLR.STRVLAN for the relevant queue.
2. By setting the CTRL.VME bit.

pinealservo, 2016-09-12:
Hmm, the big question is *why* the VLAN tags got stripped. I don't believe that's normal behavior for the igb_avb driver.

Also, the talker's Q-tag information isn't correct for the default VLAN, but I suspect that's because simple_talker and simpler_listener seem to be set up with the assumption that you've got an AVB-capable switch between them; neither will declare the default SRP Domain attribute without hearing it from their peer first.

It might help if you gave us a bit more information about your setup:

1. Network topology - are your endpoints back-to-back or through a switch? If through a switch, is it AVB-aware? List the model if you're able to.
2. Endpoint configuration - What hardware? OS? OpenAVB version--i.e. which branch, and how up-to-date is it?
3. Commands used to load drivers, launch daemons, etc.

eric-mann, 2016-09-12:
The most direct method is to run ethtool -k/ethtool -K to get/set whether VLAN offload is enabled or not. ... I can't recall whether VLAN offload is a default option, or if it is if you create a VLAN on the interface or not, or some kernel-option specific setting. 

I do know the igb_avb driver differs from the standard igb driver by allowing receives of unknown VLANs - the standard igb driver filters out packets with unknown VLANs. But since 'avb' networks are fond of mixing tagged and untagged (aka best-effort) traffic on the same port, the igb_avb driver disables the filtering rule to make it easier for users.

dmlarson4, 2016-09-14:
@eric-mann Thanks for pointing me to CNTL.VME bit! I was able to negate the bit to disable the 802.1Q Tags from getting stripped. It seems like the I210 card I'm using has this bit enabled on start up. Thanks for the timely response!

closed 2016-09-14 by dmlarson4