Avnu/OpenAvnu #481: Instantiating two talkers and two listeners

dmlarson4, 2016-09-15:
Starting with simple_rx and simple_talker as my base, I’ve added code which now instantiates two talkers and two listeners. I’ve verified one talker and one listener. The talkers and listeners are able to successfully find eachother and pair. However when I print the contents of a packet received by a listener, it prints the correct number of bytes but each byte is zero. This occurs for both listeners.

This leads me to believe that in igb_receive(), curr_pkt->vaddr is not pointed to the correct spot in memory. I’m a little bit confused how our receive buffers are actually set up and how the DMA is interacting with the software. 

Could you provide some insight into igb.c or point me to documentation that explains this layer?

Thanks,
dmlarson4

pinealservo, 2016-09-19:
Sorry, there isn't much in the way for documentation of igb.c right now. 

The gist of it (from memory--don't rely on any specifics that follow) is that igb.c is a userspace network driver component that knows how to drive some of the hardware transmit and receive queues of an Intel i210 NIC in coordination with the igb-avb kernel driver. There's an interface with the kernel driver that allows it to get references to kernel-allocated memory for buffers as well. Those get programmed into descriptors and added to the transmit and receive rings. The normal mode of operation is to busy-loop waiting for the descriptors to get filled out by the hardware and then process the buffers referenced by them.

You haven't provided much detail about your hardware or software configuration, so I don't have much to go on to speculate as to what isn't working.

Common trouble points:

1. More than one i210 NIC in a machine
2. Incorrect igb driver loaded
3. Trying to do more than one stream on a machine with simple_talker/simple_listener/simple_rx

Useful information:

1. What's your hardware configuration?  Include as much information as possible about PC hardware, NICs, network topology, AVB-support status of any other network hardware, etc.
2. What's your software configuration? OS version, branch of Open-AVB code, which daemons are running and what command lines used to launch them, etc. Include any terminal output.
3. Any other diagnostic information, like packet sniffer traces or debug code output.

eric-mann, 2016-09-20:
I would usually say 'can you get a pci trace' ... but that might be a bit much to ask ... 

From your description, I'm suspicious of your initialization path, like perhaps the rx descriptor ring base pointer was set, but the rx desc ring itself (which has the buffer pointers) was overwritten.  For example, I don't know the behavior of the device is specified if you program the rx desc base pointer, start the ring, then try to overwrite the rx desc base pointer while the queue is active. GENERALLY, I believe the write would be thrown away or not relatched (as general design practice is for hw to keep an internal context of the queues rather than use the registers).

In this case, the rx descriptors would appear to be updated correctly, but the packet data is writtent o packet buffers written 'somewhere else'. 

To confirm / deny this theory, you can periodically dump the buffer addresses in each rx descriptor to verify the address in the descriptor matches what you think is posted to the device.

fwiw.