Avnu/OpenAvnu #482: Which demon responsible for Traffic shaping? There are only 3 demons availbale - maap, mrpd, ptp.

sachindes, 2016-09-16:


pinealservo, 2016-09-19:
Traffic shaping responsibilities are split between SRP and AVTP Talkers. Each Talker is responsible for declaring the correct TSpec information in its Talker Advertisement SRP Attribute, and then for correctly shaping its own stream transmissions. Each endpoint with Talkers is then responsible for aggregating the reservations in each SR Traffic Class (which can be learned from SRP) and policing the egress queues for each SR Class so that they do not exceed their bandwidth.

Code-wise, the MRP daemon aggregates information about declared attributes, and you can use its socket interface to query the attributes that Talkers have declared. There is no standard way in Linux to configure Qav traffic shaping yet, but the igb library has some functions that AVTP Talkers can use to shape their traffic. The avtp-pipeline code makes an attempt to shape its traffic correctly, but doesn't currently work very well in a desktop Linux environment on x86 hardware without a few adjustments.

There's work in progress to improve avtp-pipeline's out-of-the-box traffic shaping behavior.

sachindes, 2016-09-20:
Thanks for info.

closed 2016-09-20 by sachindes