Avnu/OpenAvnu #484: On the XMOS board to play the video problem

jluhuangj, 2016-09-19:
I want to play video between the two boards,, but XMOS board is not original LCD, I would like to ask is that the board directly to the outside of the LCD screen can do? Has anyone done the same thing?
Thank you for spending time to watch, and my English is not good I am sorry.

https://www.xmos.com/support/boards?product=14769

andrew-elder, 2016-09-19:
I don't think XMOS example code supports video transport over AVB. You would have to do something custom.

pinealservo, 2016-09-19:
That XMOS board doesn't have any LCD screens either, and I don't know if there are enough resources left on the chip to do much in the way of video if there were an LCD or camera.  A quick look at their AVB/TSN page doesn't show any video-related hardware or code either.

There are some Freescale/NXP boards that have been used for automotive camera over Ethernet applications, but I've never done video with them myself. My only video AVB experience has been with FPGA-based solutions that we developed largely in-house.

andrew-elder, 2016-09-22:
Closing because the question has been answered.

closed 2016-09-22 by andrew-elder