Avnu/OpenAvnu #488: Can we install PTP support for i210 card with virtualized linux( use of virtual machine)?

sachindes, 2016-10-05:
Is it possible to enable kernel support for PTP which is necessary for gPTP of IGB for Open AVB within virtualised linux. Or it we mush have host OS as linux( with PTP support enabled)  and then only we can continue gPTP/PTP?

intel-ethernet, 2016-10-05:
Please define “virtualized linux”. There’s only about 4+ variations with different implications for gPTP. And is linux the guest or the host?

sachindes, 2016-10-07:
My query is for scenario where Linux is guest OS.  Can we use demonstrate Open AVB examples( Simple talker/Listener) in a setup where Linux is guest OS. It seems for HW timestmap, Linux should not be guest OS.

pinealservo, 2016-10-10:
Saying "where Linux is guest OS" still leaves a lot of variables undefined that may be necessary to give a definitive answer, although I'm not conversant enough in the area of virtualized networking to say which ones specifically are relevant.  But I'll list some likely ones for you:

1. Which virtualization technology are you wanting to use? The specifics could be important.
2. Using a virtualized network device or some form of PCI passthrough for exclusive control of a host NIC from a guest OS?
3. What is the host OS and what is the guest OS?
4. Do you want to know what's possible with today's virtualization hardware/software support or what's potentially possible with some software/driver changes?
5. Do you want to have multiple guests accessing the same host hardware device with timestamping capability? If so, what sort of virtual network topology are you expecting?

Just thinking about those questions and going through the manuals/support forums for the relevant technologies ought to get you started on an answer, but if you don't find one you can try asking a more specific question here.

sachindes, 2016-10-12:
@ pinealservo
1.	Which virtualization technology are you wanting to use? The specifics could be important.
	Oracle Virtual Machine Virtual Box with Windows 7 as Host/native OS and Linux( Ubuntu) as guest OS.
2.	Using a virtualized network device or some form of PCI pass through for exclusive control of a host NIC from a guest OS?
	Yes, Is this possible?
3.	What is the host OS and what is the guest OS?
	Windows 7 as Host/native OS and Linux( Ubuntu 14.04) as guest OS.
4.	Do you want to know what's possible with today's virtualization hardware/software support or what's potentially possible with some software/driver changes?
	Oracle Virtual Machine Virtual Box with Windows 7 as Host/native OS and Linux( Ubuntu) as guest OS.
5.	Do you want to have multiple guests accessing the same host hardware device with timestamping capability? If so, what sort of virtual network topology are you expecting?
	Yes, Guest should access PCI network HW device’s timestamp capability. But I am checking of possibility of single Guest OS.

Yes basically I was trying Open AVB simple talker/listener example demonstration with i210 card onto 2 PCs via Linux installed as GUEST OS.

Queries or issues specified in forum doesn’t specify where practical’s done with Open AVB are based on Native Linux OS or Linux as Guest OS.

pinealservo, 2016-10-17:
PCI passthrough is not available on Windows hosts for VirtualBox.

It's theoretically possible for a virtualized network device to support a hardware timestamp interface and implement it via real network device hardware timestamping, but I don't believe it's currently implemented anywhere.

Your desired configuration isn't possible right now, and isn't likely to be possible any time soon.

You might consider a live-media Linux distribution if you can't install Linux on these machines. This lets you boot into Linux from a USB drive or CD-ROM without disturbing the permanently-installed OS already on the machines.

closed 2016-10-17 by pinealservo

sachindes, 2016-11-22:
Thanks. I had moved onto native Linux machine.

sachindes, 2019-01-22:
@pinealservo ,

In my recent work, I could use Linux VM and existing NIC card (not i210) and could stream AVB audio with AVNu stack.
I need to configure AVB in static way, PTP may work for SW time stamp (i had skipped it).
However AVB packets received and played on Linux VM's Audio output.
With this I could test AVB streaming on my Laptop.

pinealservo, 2019-01-23:
This will not work. It will likely never work. If you don't understand why I'm saying it won't work, please look more closely into the actual requirements for AVB endpoint behavior. It's a lot more involved than just unpacking audio samples from packets and shoving them into ALSA. That's just the easy part.

sachindes, 2019-01-24:
Yes, agree it's unpacking and playing with alsa without any daemons and HW
time stamp. However it help to verify AVTP without need of special setup.
Actual test would involve a valid end point.