Avnu/OpenAvnu #493: gPTP : PTPMessagePathDelayRespFollowUp::processMessage incorrect?

wmoors, 2016-10-06:
802.1AS clause 11.2.15.2.4 describes the computation of the link delay as follows...

link_delay = r * ( t<sub>4</sub> – t<sub>1</sub> ) – ( t<sub>3</sub> – t<sub>2</sub> ) / 2  whereas t<sub>3</sub> is described as 

> t<sub>3</sub> = sum of (1) the ns field of the responseOriginTimestamp, (2) the seconds field of the
responseOriginTimestamp multiplied by 10<sup>9</sup> , and (3) the correction field divided by 2<sup>16</sup> (i.e.,
the correction field is expressed in ns plus fractional ns), of the Pdelay_Resp_Follow_Up
message received in response to the Pdelay_Req message sent by the MD entity

However, when looking into the code (`PTPMessagePathDelayRespFollowUp::processMessage`), I cannot find any reference to the correction field of the PDelay_Resp_Follow_Up message.

Secondly, it also seems like it is not (t<sub>4</sub> - t<sub>1</sub>) that is multiplied by r (the `neighborRateRatio`), but instead (t<sub>3</sub> - t<sub>2</sub>) is.

Could someone confirm this?

wmoors, 2016-10-06:
Perhaps a little background. I was trying to get 2 gPTP daemons (running on I210 cards) to sync through a switch we have in development. 

So I cross checked with the 1588 specification and noticed that PTPv2 allows for 2 different ways of formatting the PDelay_Resp_Follow_Up message.
Our implementation picked the one where the complete `pdelayRespEventEgressTimestamp` is put in the correction field of the PDelay_Resp_Follow_Up. So the link delay when exchanged with our switch is off the charts... :/

pinealservo, 2016-10-10:
Could you supply some more information about which branch and, if possible, which git ref you were building against?  There were some fixes in the last couple of months regarding some calculations in this general area, and I'm not sure if they were merged to the master branch or not before a few days ago.

I did a deep review of the rate ratio calculation code at the time the changes I'm thinking of were made, but it would take some time to refresh my memory of the relevant context, and I'd like to make sure that I'm looking at the same code you are and that it's the most recent version before I dive into it again.

andrew-elder, 2016-10-28:
@wmoors - is the company you work for a member of Avnu?

wmoors, 2016-11-02:
@andrew-elder : yes. But I don't think I'm allowed to disclose the company. 

@pinealservo : I'm building open-avb-next : 17fdab7 

I have rewritten the link_delay mechanism in our code and gPTP seems to be working now.

andrew-elder, 2016-11-02:
@wmoors - I don't think there is any requirement to publically disclose who you work for. Can you email me at aelder@audioscience.com ?

andrew-elder, 2016-11-02:
@wmoors - are you able to provide a pull request with your changes in them so that others can easily review your updates?

andrew-elder, 2016-11-10:
This issue was reviewed today on an Avnu conference call. Current gPTP code was confirmed to be functionally correct. An additional comment block will be added a future date. One thing that differs from 802.1AS is that the link_delay variable is stored in terms of local clock time, whereas the spec defines how to compute in in GM clock time units.