Avnu/OpenAvnu #494: OpenAVB over normal Switch-Device

pwosia, 2016-10-10:
Hi, does someone has experience in setting up an OpenAVB Environment on a normal Switch (e.g. Cisco SG-300 ). I am aware about using AVB-Switch but I would like to try alternatives.
My test-lap consists of 2 Ubuntu 12.04 machines with Intel i210 NIC connected over a Cisco SG-300 Switch.
The PC can ping each other. But if the gptp-daemon starts, then the Sync-Threshold is too big and each PC takes it clock as Grand-Master-Clock. 
Is there a way to adjust the switch-behavior or do I have to stick to AVB-Switch-Device?

pinealservo, 2016-10-10:
The AVB specifications are designed to specifically disallow non-AVB bridges, as they are not capable of providing the traffic shaping services that AVB's guarantees rely on.

You can get around this for testing/experimentation purposes by increasing the `neighborPropDelayThresh` parameter in the `gptp_cfg.ini` configuration file to something beyond the measured values you're seeing, but please understand that this isn't how AVB is meant to work and you'll be largely on your own as to figuring out how to get anything useful out of this configuration.

andrew-elder, 2017-02-13:
Question answered.

closed 2017-02-13 by andrew-elder