Avnu/OpenAvnu #497: SRP End to End Response Time

deshpanv, 2016-10-13:
Hello all,

I am new to Open AVB and related stuff. I want to calculate end to end response time for a packet over AVB Network as a function of Class and Priority.

I am successfully able to calculate end to end response time over non avb network. Now I just want to reserve a stream and then send packet to calculate end to end response time and taking results for class A, Class B with different Priorities. 

To do this, I looked on the code of simple talker and listener, I tried to run them and it is working for me but I am unable to locate how does the reservation of stream takes place and how to change the attributes like class/priority.

I am looking for hint on how to proceed further to change the attributes and reserve stream and send my packet so that I can calculate end to end response time.

Thank you in advance for the help and guidance.

pinealservo, 2016-10-17:
The code to configure the i210's hardware traffic shaping capability is in the `lib/igb` directory. The `igb_set_class_bandwidth` function call in simple_talker is how that's done. If your network consists of standards-compliant AVB-capable hardware, you can set up the shaping for the stream throughout the rest of the network with the SRP protocol.

Look at `examples/common/talker_mrp_client.c` for the code that talks to the MRP daemon to establish a stream reservation.  There's code calling several of those functions in simple_talker and if things are set up right, they *should* make a stream reservation.

I'm not sure exactly what you're trying to show with your tests here, but when you say "end to end response time" it sounds like you're talking about a round trip for a request/response pair, which is not really what AVB is designed for. AVB ensures that isochronous unidirectional streaming traffic gets from talker to listener with minimal disturbance *in the presence of interfering traffic* on the network. The various knobs you can adjust regarding priority and traffic shaping only affect queuing behavior, and if talkers are behaving properly, queues throughout the network will never even start to fill up unless there's a bunch of other traffic. AVB ensures that the stream traffic goes out first, but gets spaced evenly so other traffic isn't completely blocked and it doesn't create a bunch of queueing in AVB queues due to bursts.

deshpanv, 2016-10-18:
Thank You very much for this precious information. 
Also, in my question, when I say End to End Response time, it means

I send a packet from one PC to another PC through AVB Switch and calculate the time the packet takes going from one PC to another.

I want to evaluate the performance of this by sending the packet with Class A Priority on reserved stream and then with Class B. I then, at the end want to compare and infer is it really makes difference to use different classes when the network is not loaded with respect to the time packet takes going from one pc to another.

But, reading you answer it is apparent that it will not make a difference unless the network is fully loaded. Can you please correct me if I am wrong ?

Thank You
Varun

pinealservo, 2016-10-18:
In an unloaded network, streaming traffic classes should have exactly the same end-to-end transit time as best-effort traffic taking the same route through the network. You can't speed up transit time with a traffic shaper; in fact, the effect of a traffic shaper is to *delay* some queued frames.

It's the combination of prioritization and shaping that keeps AVB traffic *consistent*, which means that the *variance* of the transit times is reduced. With a low variance, you can assign with confidence a smaller buffer size, and thus (maybe somewhat counterintuitively) reduce your overall latency by *slowing down* some of your stream's packets when they're bunched up by interfering traffic.

deshpanv, 2016-10-31:
Hello,

I studied the program of simple talker, in that the program is calling `mrp_register_domain()` but the domain is not changed when I recall `mrp_get_domain()`. When I digged deep I found that mrpd client is not doing anything for commands starting with S. I want to change domain for the stream I am sending through talker but I am unable to do it. Can you please guide me in this regards ?

andrew-elder, 2016-10-31:
Are you able reference the module and line number that led you to conclude that commands started with S are ignored by mrpd? There is test code https://github.com/AVnu/Open-AVB/blob/master/daemons/mrpd/tests/simple/msrp_tests.cpp#L600 that sends strings starting with "S" into mrpd code and those tests seem to work ok.

deshpanv, 2016-10-31:
Hello,

In the `mrpd.c` file, 

This is the code 

```
switch (buf[0]) {
	case 'M':
		printf("Command :  %s\n", buf);
		return mmrp_recv_cmd(buf, buflen, client);
		break;
	case 'V':
		printf("Command :  %s\n", buf);
		return mvrp_recv_cmd(buf, buflen, client);
		break;
	case 'S':
	case 'I':
		printf("Command :  %s\n", buf);
		return msrp_recv_cmd(buf, buflen, client);
		break;
```

As you can see, for the case S, it is left blank. But when  we want to change any attribute of the stream, the command starts with S. So I concluded that it is not doing anything. Can you please correct me if I am wrong ?

pinealservo, 2016-10-31:
In C, cases have fall-through behavior by default. Any `case` label without a `break` will have execution continue into the next `case`. So, in this situation, the last bit of code applies to commands starting with both `S` and `I`.

andrew-elder, 2016-11-01:
@deshpanv - can we close this issue then?

deshpanv, 2016-11-03:
Hello,

Can you please guide me what changes I should make to simple talker and listener so that the AVB Communication is done on Class B.

Also how can I change priority for the classes. I have edited the code by myself but it doesn't seem to work. I thank you for the guidance !

andrew-elder, 2016-11-03:
Good policy is to provide as detailed a list as possible of what you have changed! That way others know where to start with there replies.

deshpanv, 2016-11-04:
Hello,

Following are the changes I made to be able to send packets using Class B 

File : https://github.com/AVnu/Open-AVB/blob/master/examples/simple_talker/simple_talker.c

```
Line 440 : 	rc = mrp_register_domain(class_b, ctx);
Line 446 :        rc = mrp_join_vlan(class_b, ctx);     
Line 452 :        if( transport == 2 ) {
		igb_set_class_bandwidth
			(&igb_dev, 0, 125000/L2_PACKET_IPG, 0, PKT_SZ - 22);
	} else {
		igb_set_class_bandwidth
			(&igb_dev, 0, 1,
			0,  sizeof(*l4_headers)+L4_SAMPLES_PER_FRAME*CHANNELS*2);
}

Line 502 :      ((ctx->domain_class_b_priority << 13 | ctx->domain_class_b_vid)) >> 8;
		((char *)tmp_packet->vaddr)[15] =
((ctx->domain_class_b_priority << 13 | ctx->domain_class_b_vid)) & 0xFF;
```

File : 

https://github.com/AVnu/Open-AVB/blob/master/examples/simple_listener/simple_listener.c

```
Line 236 : rc = report_domain_status(class_b,ctx);
Line 242 : rc = join_vlan(class_b, ctx);

```

But it doesn't seems to work. I get askready state on talker side but I am unable to receive packets on listener's side.

In wireshark, I am unable to see AVB packets on listener's side.
For changing the Priority, I have inserted following code in talker application.

```
rc = mrp_get_domain(ctx, class_a, class_b);
	if (rc) {
		printf("failed calling msp_get_domain()\n");
		return EXIT_FAILURE;
	}
	printf("detected domain Class A PRIO=%d VID=%04x...\n",class_a->priority,
	       class_a->vid);
        // class_a->priority = class_a->priority - 1;
        class_a->priority = class_a->priority + 1;
	rc = mrp_register_domain(class_a, ctx);
	if (rc) {
		printf("mrp_register_domain failed\n");
		return EXIT_FAILURE;
	}
```

But on wireshark I see the Priority 3 only even if I change it.
I also want to know as the priority field is 3 bit. So, are priorities linked to classes ?
For example : 
Priority 0 to 3 == Belongs to Class A
Priority 4 to 7 == Belongs to Class B

Because when I analyze AVB Packet, Only priority and vid are sent with each packet but it is never mentioned in avb packet to which class it belongs ? I am little confused with priority and class link.

Thank  you for the help.

pinealservo, 2016-11-04:
This comment I left on another issue might help you understand how the priority field is used: https://github.com/AVnu/Open-AVB/issues/470#issuecomment-245367253

SRP can be used to change the priority to traffic class mapping.

deshpanv, 2016-11-04:
Hello,

You mean to say that in the AVB network, if you send a packet with priority 2, then it is automatically associated with class B and with priority 3, it is associated to class A.

And can I assign Priority 2 with Class A ? Also, can I assign Priority 2 to both class A and B ?

Is it possible to assign all priorities to class A or class B or both ?

Example Priorities 0-7 can be mapped with class A alone ?

pinealservo, 2016-11-04:
SRP Domain attributes show the map of priority to SR Class.  AVB switches come with default SRP Domain attribute values; they don't propagate, so they have to be set via some manual configuration if you want to change them from their defaults, and they'll have to be changed identically throughout the network. You have to use the same SRP Domain attribute values on the endpoints as the switches are declaring.

This means that in order to use Class B, you need to see what (if anything) the switch is declaring for Class B in its domain attribute declarations. The code in simple_talker.c calls `mrp_get_domain()` and prints out the information for class A; you should have it print out class B instead to ensure it actually supports Class B. The changes you made *should* make your class B declaration match the switch's declaration, but you should verify this at the switches to ensure that the domain boundaries for class B are set where you expect them to be.

You must use the priority value for class B that's declared by the switch; this has to be used both in the talker advertise attribute for the stream as well as the packet data itself. You'll also need to ensure you use the right parameters to igb_set_class_bandwidth() and igb_xmit(). I believe one of the parameters selects which hardware queue to use, and you have to set the bandwidth on the same queue that you transmit on. I don't know much about how the igb functions work, though, so you may want to look at the igb library source.

deshpanv, 2017-01-17:
Thank You for the help, it solved my issue. I have another issue which I opened, Can you please guide me with that ?

https://github.com/AVnu/Open-AVB/issues/541

closed 2017-01-17 by andrew-elder